alibaba/AliOS-Things #878: alios 如何调用 linkkit sdk的

scarleast, 2019-02-26:
**Describe the bug**

**environment**
1. branch(version): rel2.0
2. Desktop OS(windows linux or macOS):Mac
3. example: 

想问下，linkkitapp的示例里面，/AliOS-Things/app/example/yinerda_mixchip_kit_led/app_entry.c并没有调用任何/AliOS-Things/app/example/yinerda_mixchip_kit_led/linkkit_example_solo.c中的任何函数，那么linkkit_example_solo.c中的代码是如何跑起来的？是在哪里调用了linkkit_example_solo.c中的代码。

xiaowenxia, 2019-02-27:
在app_entry.c 中的`wifi_service_event()`函数中调用的：

```c
#ifdef MQTT_DIRECT
        aos_task_new("linkkit", (void (*)(void *))linkkit_main, NULL, 1024 * 6);
#else
        aos_task_new("linkkit", (void (*)(void *))linkkit_main, NULL, 1024 * 8);
#endif
```

https://github.com/alibaba/AliOS-Things/blob/rel_2.0.0/app/example/linkkitapp/app_entry.c#L64

scarleast, 2019-02-27:
好的，感谢

closed 2019-02-27 by scarleast