alibaba/AliOS-Things #879: MQTT Payload 不被被阿里云端识别

wizephen, 2019-02-27:
ALiOS Things Version 2.0.0

问题描述：
1.  MQTT demo 发送 payload内容时，云端设备控制中心： 日志服务-》状态及与原因分析 显示 
“失败：No authorization”
[image]

2.  可以确定，阿里云端已经成功识别到设备，因为，云端显示 设备 "在线";
[image]

3.  可以确定，设备端 developerkit 已经正常工作，并loop打印的log 如下(MQTT  payload  内容为： "{\"attr_name\":\"temperature\", \"attr_value\":\"%d\"}"， 代码中也可以直接找到)， 我认为MQTT Push 是成功的；

_event_handle|090 :: publish success, packet-id=5
[dbg] MQTTPublish(303): WRITE COMPLETED: curr buf_send = 0x200126e0, curr buf_size_send = 1024
mqtt_client|267 :: packet-id=6, publish topic msg={"attr_name":"temperature", "attr_value":"1"}
[dbg] iotx_mc_cycle(1663): PUBACK
event_handle|090 :: publish success, packet-id=6
[dbg] MQTTPublish(303): WRITE COMPLETED: curr buf_send = 0x200126e0, curr buf_size_send = 1024
mqtt_client|267 :: packet-id=7, publish topic msg={"attr_name":"temperature", "attr_value":"1"}
[dbg] iotx_mc_cycle(1663): PUBACK
event_handle|090 :: publish success, packet-id=7_

4.  云端，我定义的产品功能如下：
[image]

5. 我有尝试将MQTT Payload 换成 物模型 的格式，其结果是push 失败，如下log 所示：

_[err] MQTTSerialize_publish is error, len=-2, buf_size_send=1024, payloadlen=984
[err] MQTTPublish is error, rc = -21
mqtt_client|267 :: error occur when publish
mqtt_client|271 :: packet-id=4294967275, publish topic msg={"schema":"https://iotx-tsl.oss-ap-southeast-1.aliyuncs.com/schema.json","profile":{"productKey":"a1jEgKrd5mb"},"services":[{"o
[err] MQTTSerialize_publish is error, len=-2, buf_size_send=1024, payloadlen=984
[err] MQTTPublish is error, rc = -21
mqtt_client|267 :: error occur when publish_

ALiers:  请帮忙分析，谢谢!

**PS：  ALiOS Thing 相关文档，个人认为很不系统！  开发者 不能根据 example 下的 readme , 快速上手， 希望这方面有所改进！**

xiaowenxia, 2019-02-28:
mqttapp需要配合物联网平台的**基础版**产品使用，如果是用的高级版的话，会出现`No authorization`，你看看是不是这个问题？

> 
PS： ALiOS Thing 相关文档，个人认为很不系统！ 开发者 不能根据 example 下的 readme , 快速上手， 希望这方面有所改进！

谢谢你的建议，以后example下面会有readme说明文档的。

wizephen, 2019-02-28:
@xiaowenxia 我在云平台上建立**基础版** 产品及对应的设备。developerkit 正常运行后，只能在云端平台检测设备**在线**， 但仍然有  ”**失败：No authorization**” 这样的信息。

closed 2022-03-01 by skylarCai