alibaba/AliOS-Things #897: aos monitor中无法键入指令

Jerry280989322, 2019-03-12:
**Describe the bug**
在进行MQTT上传数据到IOT套件的过程中，mqttapp烧录成功后，链接developerkit，但是链接后不能键入任何字符，只能按 “ctrl+组合键”查看menu等，求问如何输入指令呢？（我想键入连接wifi的指令：netmgr connect  *****   *****）之前莫名其妙成功了两次，但是现在又不行了，自己也搞不清原因。

**environment**
1. branch(version): master
2. Desktop OS(windows linux or macOS):windows 10
3. example: mqttapp

xiaowenxia, 2019-03-12:
@Jerry280989322 需要确认一下developerkit是否正常运行。developerkit有时候烧录完成之后，需要插拔一下或者重新烧录才能正常运行的。

Jerry280989322, 2019-03-12:
> @Jerry280989322 需要确认一下developerkit是否正常运行。developerkit有时候烧录完成之后，需要插拔一下或者重新烧录才能正常运行的。

嗯嗯，刚刚按你说的试了一下，可以了~  Thanks~  不过我试过很多次，每次烧录之后都得插拔一下并且按一下reset键才能正常键入指令，这个问题能解决吗？

xiaowenxia, 2019-03-12:
@Jerry280989322 developerkit目前确实存在这个问题，你可以到[诺行开发板钉钉群](https://i.imgur.com/rqjjjA0.jpg)里面了解到最新的动态。

Jerry280989322, 2019-03-12:
> @Jerry280989322 developerkit目前确实存在这个问题，你可以到[诺行开发板钉钉群](https://i.imgur.com/rqjjjA0.jpg)里面了解到最新的动态。

好的，谢谢

closed 2019-03-12 by xiaowenxia

zeng-hello, 2019-08-12:
@xiaowenxia 你好，我在重新插拔并reset后，还是不能在aos monitor 下输入命令，只能ctrl+]  退出uu才能输入netmgr connect.但是系统提示错误.
如下：
zeng@zeng-virtual-machine:~/tool/AliOS-Things$ aos monitor /dev/ttyS0 115200
--- Miniterm on /dev/ttyS0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
--- unknown menu character Ctrl+J --

--- exit ---
zeng@zeng-virtual-machine:~/tool/AliOS-Things$ netmgr connect 1234 123456781
netmgr：未找到命令
zeng@zeng-virtual-machine:~/tool/AliOS-Things$