alibaba/AliOS-Things #901: 2.1.0编译不过,求帮看

xiake468, 2019-03-13:
**Describe the bug**

process_begin: CreateProcess(NULL, dd if=/dev/zero of=out/a_yang_prj@mk3060/binary/a_yang_prj@mk3060.tmp.bin bs=28 count=1, ...) failed.
make (e=2): 系统找不到指定的文件。
make.exe[1]: *** [platform/mcu/moc108//gen_crc_bin.mk:41: gen_crc_bin] Error 2
make.exe: *** [build/Makefile:395: main_app] Error 2
[AliOS-Things] ERROR: "E:/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/AliOS-Things/build -e -f build/Makefile" in "E:\AliOS-Things"
---
mk3060板子,写好程序,和Config.in,编译之后出错误没有ota.bin文件生成呢,求帮忙分析以下是哪里除了问题,2.1.0的sdk

**environment**
1. branch(version): 
rel2.1.0
2. Desktop OS(windows linux or macOS):
win10 63bit
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

xiake468, 2019-03-13:
config.in的内容
config AOS_APP_A_YANG_PRJ
    bool "a_yang_prj"
    select AOS_COMP_CLI 
    select AOS_COMP_NETMGR
    select AOS_COMP_CJSON
    select AOS_COMP_LINKKIT_SDK_C
    select APP_FEATURE_LINKKIT
aos.mk内容
$(NAME)_SUMMARY := a_yang_app

$(NAME)_SOURCES += main.c \
				   ts_jdq.c \
				   ts_key.c \
				   ts_wifi.c \
				   ts_linkkit.c 

$(NAME)_COMPONENTS += linkkit_sdk_c \
					  netmgr \
					  cjson \
					  cli

GLOBAL_INCLUDES += ./

代码应该不用贴了吧...其实就是linkkit_app改成我自己项目用的

xiaowenxia, 2019-03-14:
@xiake468 需要配合使用git bash终端来编译哈，或者使用linux编译也可以。

xiake468, 2019-03-14:
> @xiake468 需要配合使用git bash终端来编译哈，或者使用linux编译也可以。

Build complete: a_yang_prj@mk3060
Making config file for first time
PLATFORM_FULL mk3060
2boot_version:aos-2nd_boot-1.0.0
all components: mcu_moc108 board/mk3060 ota_2nd_boot
2boot_version:aos-2nd_boot-1.0.0
build/build_rules/aos_target_config_2boot.mk:321: *** No application specified. Options are: uart udataapp http2app id2_app yts linkkitapp udevapp modbus_demo bluetooth README.md ulocationapp das_app mqttapp linkkit_gateway meshapp otaapp tinyengine_app coapapp athostapp tls blink Config.in itls_app helloworld sk dk lorawan prov_app.  Stop.
make.exe: *** [build/Makefile:401: out/a_yang_prj@mk3060/config_2boot.mk] Error 2
Check if required tools for mk3060 exist
[AliOS-Things] ERROR: "E:/alios_thing/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/alios_thing/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/alios_thing/AliOS-Things/build -e -f build/Makefile" in "E:\alios_thing\AliOS-Things"
---这是在win10的gitbash的编译结果,还是编译不出ota.bin呢,等下我弄个unbuntu的虚拟机再试一下

xiake468, 2019-03-14:
> @xiake468 需要配合使用git bash终端来编译哈，或者使用linux编译也可以。

还想问一下,我的项目中没弄那个ucube.py的文件,不知道是不是这个的锅,因为还不太理解这个文件是干嘛的,是scons的那个配置文件吗

xiaowenxia, 2019-03-14:
@xiake468 你这样：
```sh
# 清除.config 和out/
aos make distclean

#配置
aos make a_yang_prj@mk3060 -c config

#编译
aos make
```

再试试能不能编译。更多指令说明参考[这个](https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-uCube.zh)。

xiake468, 2019-03-14:
还是不行哦,window用gitbash还是同样的结果,ubuntu虚拟机aos make a_yang_prj@mk3060 -c config这条命令报了错误

> @xiake468 你这样：
> # 清除.config 和out/
> aos make distclean
> 
> #配置
> aos make a_yang_prj@mk3060 -c config
> 
> #编译
> aos make
> 再试试能不能编译。更多指令说明参考这个。

ma77045728@ubuntu:/mnt/hgfs/ubuntu18_04_share/alios_210/AliOS-Things$ aos make distclean
aos-cube version: 0.3.1
Cleaning...
Done
ma77045728@ubuntu:/mnt/hgfs/ubuntu18_04_share/alios_210/AliOS-Things$ aos make a_yang_prj@mk3060 -c config
aos-cube version: 0.3.1
Check if required tools for mk3060 exist
fatal: destination path './build/kconfig' already exists and is not an empty directory.
build/build_rules/aos_kconfig.mk:60: recipe for target 'build/kconfig/kconfig-conf' failed
make: *** [build/kconfig/kconfig-conf] Error 128
[AliOS-Things] ERROR: "/mnt/hgfs/ubuntu18_04_share/alios_210/AliOS-Things/build/cmd/linux64/make" returned error code 2.
[AliOS-Things] ERROR: Command "/mnt/hgfs/ubuntu18_04_share/alios_210/AliOS-Things/build/cmd/linux64/make HOST_OS=Linux64 TOOLS_ROOT=/mnt/hgfs/ubuntu18_04_share/alios_210/AliOS-Things/build -e -f build/Makefile a_yang_prj@mk3060.config" in "/mnt/hgfs/ubuntu18_04_share/alios_210/AliOS-Things"
---

xiaowenxia, 2019-03-14:
@xiake468 你删除`build/kconfig` 目录再试试吧。

WenzongFan, 2019-03-14:
@xiake468 Config.in文件添加可以参考[这里](https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Build-Configuration.zh)，添加App和board通常需要更新两个地方：

xiake468, 2019-03-14:
> @xiake468 Config.in文件添加可以参考这里，添加App和board通常需要更新两个地方：1 app自己的Config.in文件 2 上一级的Config.in文件用来包含app的Config.in

是这样的...这个我有看的,我的项目目录叫a_yang_prj,放在了app目录中(和example同级目录),app目录中的config.in我改成了
config AOS_APP_LOCALAPP
    bool "Local Applications"

if AOS_APP_LOCALAPP

### Sample for user's application:
### The app's Config.in must be there before "source" it
#
choice
    prompt "Select application"
#
source "app/a_yang_prj/Config.in"
if AOS_APP_A_YANG_PRJ 
    config AOS_BUILD_APP
        default "a_yang_prj"
endif
#
# source "app/myapp2/Config.in"
# if AOS_APP_MYAPP2
#     config AOS_BUILD_APP
#         default "myapp2"
# endif
#
endchoice

endif
然后我自己项目的目录a_yang_prj中的config.in是这样的

config AOS_APP_A_YANG_PRJ
    bool "a_yang_prj"
    select AOS_COMP_OSAL_AOS 
    select AOS_COMP_CLI 
    select AOS_COMP_NETMGR
    select AOS_COMP_CJSON
    select AOS_COMP_LINKKIT_SDK_C
    select APP_FEATURE_LINKKIT

    help
        a_yang_prj

if AOS_APP_BLINK
# Configurations for app blink
endif    
目前问题还是没解决,怎么弄都不行,最终的结果都是  
Build complete: a_yang_prj@mk3060
Making config file for first time
PLATFORM_FULL mk3060
2boot_version:aos-2nd_boot-1.0.0
all components: mcu_moc108 board/mk3060 ota_2nd_boot
2boot_version:aos-2nd_boot-1.0.0
build/build_rules/aos_target_config_2boot.mk:321: *** No application specified. Options are: uart udataapp http2app id2_app yts linkkitapp udevapp modbus_demo bluetooth README.md ulocationapp das_app mqttapp linkkit_gateway meshapp otaapp tinyengine_app coapapp athostapp tls blink Config.in itls_app helloworld sk dk lorawan prov_app.  Stop.
make.exe: *** [build/Makefile:401: out/a_yang_prj@mk3060/config_2boot.mk] Error 2
Check if required tools for mk3060 exist
[AliOS-Things] ERROR: "E:/alios_thing/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/alios_thing/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/alios_thing/AliOS-Things/build -e -f build/Makefile" in "E:\alios_thing\AliOS-Things"

> @xiake468 你删除build/kconfig 目录再试试吧。

xiake468, 2019-03-14:
> @xiake468 Config.in文件添加可以参考这里，添加App和board通常需要更新两个地方：1 app自己的Config.in文件 2 上一级的Config.in文件用来包含app的Config.in
>
> 是这样的...这个我有看的,我的项目目录叫a_yang_prj,放在了app目录中(和example同级目录),app目录中的config.in我改成了
> config [...]

大写的部分在文件中是两个井号注释,刚刚发出去被当成makedown了估计

xiake468, 2019-03-14:
前面写的太乱了,我重新描述一下问题,我的项目目录叫a_yang_prj,这个目录放在app目录中(与example目录同级),根据文档,我改写了app目录下的config.in文件,改完如下

```
config AOS_APP_LOCALAPP
    bool "Local Applications"

if AOS_APP_LOCALAPP

### Sample for user's application:
### The app's Config.in must be there before "source" it
#
choice
    prompt "Select application"
#
source "app/a_yang_prj/Config.in"
if AOS_APP_A_YANG_PRJ 
    config AOS_BUILD_APP
        default "a_yang_prj"
endif
#
# source "app/myapp2/Config.in"
# if AOS_APP_MYAPP2
#     config AOS_BUILD_APP
#         default "myapp2"
# endif
#
endchoice

endif
```
然后写了我项目目录中的config.in文件,内容如下
```
config AOS_APP_A_YANG_PRJ
    bool "a_yang_prj"
    select AOS_COMP_OSAL_AOS 
    select AOS_COMP_CLI 
    select AOS_COMP_NETMGR
    select AOS_COMP_CJSON
    select AOS_COMP_LINKKIT_SDK_C
    select APP_FEATURE_LINKKIT

    help
        a_yang_prj

if AOS_APP_A_YANG_PRJ
# Configurations for app blink
endif    
  
```
在win10的git bash,然后就执行了make a_yang_prj@mk3060 -c config,是成功的,然后执行aos make,是有结果的,下面这个东西已经出来了,报了错误,看情况应该是部分编译成功了,只是没有出来我想要的ota.bin这个文件
```
                        AOS MEMORY MAP
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| alicrypto                                | 30713     | 8752     |
| arch_armv5                               | 388       | 0        |
| a_yang_prj                               | 942       | 68       |
| board_mk3060                             | 737       | 24       |
| cli                                      | 5964      | 353      |
| debug                                    | 1519      | 0        |
| imbedtls                                 | 28813     | 76       |
| iotx-hal                                 | 5596      | 204      |
| kernel_init                              | 875       | 36       |
| kv                                       | 3403      | 36       |
| libawss                                  | 30189     | 1511     |
| libc_nano                                | 21924     | 464      |
| libdev_bind                              | 8467      | 147      |
| libdev_reset                             | 1171      | 35       |
| libgcc                                   | 6400      | 0        |
| libiot_coap_coappack                     | 1658      | 0        |
| libiot_coap_local                        | 7618      | 24       |
| libiot_log                               | 47        | 29       |
| libiot_mqtt                              | 12203     | 24       |
| libiot_system                            | 4825      | 360      |
| libiot_utils                             | 14618     | 4        |
| libmoc108                                | 208729    | 52721    |
| libprov                                  | 624       | 0        |
| ls_hal                                   | 176       | 0        |
| ls_osa                                   | 110       | 0        |
| lwip                                     | 39350     | 3295     |
| netmgr                                   | 2195      | 274      |
| newlib_stub                              | 384       | 0        |
| osal_aos                                 | 2108      | 0        |
| rhino                                    | 13625     | 10519    |
| ulog                                     | 507       | 18       |
| vfs                                      | 1184      | 1105     |
| yloop                                    | 2440      | 32       |
| *fill*                                   | 373       | 169      |
|=================================================================|
| TOTAL (bytes)                            | 459875    | 80280    |
|=================================================================|
1+0 records in
1+0 records out
28 bytes copied, 0.00175957 s, 15.9 kB/s
496326
1fa9ef78aa0b47f645afa6daa6bb7739
Build complete: a_yang_prj@mk3060
Making config file for first time
PLATFORM_FULL mk3060
2boot_version:aos-2nd_boot-1.0.0
all components: mcu_moc108 board/mk3060 ota_2nd_boot
2boot_version:aos-2nd_boot-1.0.0
build/build_rules/aos_target_config_2boot.mk:321: *** No application specified. Options are: uart udataapp http2app id2_app yts linkkitapp udevapp modbus_demo bluetooth README.md ulocationapp das_app mqttapp linkkit_gateway meshapp otaapp tinyengine_app coapapp athostapp tls blink Config.in itls_app helloworld sk dk lorawan prov_app.  Stop.
make.exe: *** [build/Makefile:401: out/a_yang_prj@mk3060/config_2boot.mk] Error 2
Check if required tools for mk3060 exist
[AliOS-Things] ERROR: "E:/alios_thing/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/alios_thing/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/alios_thing/AliOS-Things/build -e -f build/Makefile" in "E:\alios_thing\AliOS-Things"
---
```

WenzongFan, 2019-03-14:
@xiake468 先workaround一下，把你的应用放到example目录下，然后修改example/Config.in这个文件，应该就可以编了

xiake468, 2019-03-14:
> @xiake468 先workaround一下，把你的应用放到example目录下，然后修改example/Config.in这个文件，应该就可以编了

> @xiake468 你删除build/kconfig 目录再试试吧。

按楼上那哥们说的git reset了一下,然后把项目放在example目录中,改了下example/Config.in,终于成功了,谢谢各位,爱你们!