alibaba/AliOS-Things #903: 2.1.0版本stm32 keil工程下大小端异常

comporsa, 2019-03-14:
**Describe the bug**
keil工程，stm32f429项目。工程选项勾选gnu extention。发现以下这句条件编译无效
#if __BYTE_ORDER__ == __ORDER_BIG_ENDIAN__
这句会导致字节序以大端的方式执行。
解决方案：改回老版本的#if defined(REVERSED)即解决。

xiaowenxia, 2019-03-14:
@comporsa 你问题描述清楚一点：
- keil工程是用aos-cube导出的吗？
- 工程选项截图？
- 编译无效具体指？
- 这行代码在哪里？
- 老版本指的是哪个分支？

comporsa, 2019-03-14:
keil工程使用的是2.0.0的helloworld@starterkit工程移植过来的，之前的2.0.0版本下MQTTPacket.h第45行代码，用于控制MQTTHeader这个union体的大小端字节序的条件编译是#if defined(REVERSED)。当时没问题。工程选项就是在c/c++选型页下勾选了c99 mode和 gnu extension，还有one elf section per function。这个选项一直没变。

在移植到2.1.0后，发现前面说的 #if defined(REVERSED)这句在新版本里改成了#if BYTE_ORDER == ORDER_BIG_ENDIAN。由于BYTE_ORDER这个对象并未定义，所以我认为该条件并不会生效。但是在编译过程中，该条件却生效，也就是变成以大端的方式来存储数据，导致mqtt的序列化不正确。

我是如何测试出来的呢？
我将下面这个union定义拿出来，放在程序开始运行的地方以好观测。
typedef union
{
	unsigned char byte;	                /**< the whole byte */
#if defined(REVERSED)
	struct
	{
		unsigned int type : 4;			/**< message type nibble */
		unsigned int dup : 1;				/**< DUP flag bit */
		unsigned int qos : 2;				/**< QoS value, 0, 1 or 2 */
		unsigned int retain : 1;		/**< retained flag bit */
	} bits;
#else
	struct
	{
		unsigned int retain : 1;		/**< retained flag bit */
		unsigned int qos : 2;				/**< QoS value, 0, 1 or 2 */
		unsigned int dup : 1;				/**< DUP flag bit */
		unsigned int type : 4;			/**< message type nibble */
	} bits;
#endif
} MQTTHeader;
然后定义一个变量，MQTTHeader m; 令m.byte=0,然后赋值 m.bits.type=1;  输出byte的值，结果是0x01。说明是大端字节序。

comporsa, 2019-03-14:
[image]
事实上，一个stm32f1的keil5工程，只要加入这句#if \_\_BYTE_ORDER\_\_ ==\ _\_ORDER_BIG_ENDIAN\_\_，都会生效，也就是说在默认情况下这个条件是成立的。和工程选项无关，经证明，那个gnu extension并不影响。

caveguy, 2019-03-18:
这样可以的，另外一种方式：在keil工程里定义 __ORDER_BIG_ENDIAN__ = 0；__BYTE_ORDER__ = 1也可以解决

closed 2019-03-29 by comporsa