alibaba/AliOS-Things #904: 想确认下2.1.0的ota.bin

xiake468, 2019-03-14:
**Describe the bug**
我的板子是mk3060,写好程序编译出ota.bin,boot脚接地用串口烧入程序后,浮空boot脚,复位板子.程序好像没有跑起来一样,什么反应也没有,什么日志也没打出来,我想就算是程序写的有问题,也应该有一些初始化日志被打出来啊,板子是好的,我烧过2.0.0编译出来的程序是可以正常运行的,现在想确认一下,是不是2.1.0的固件烧入方式有所改变,因为我发现2.1.0之后的编译过程好像比较之前有所不同

**environment**
1. branch(version): 
2,1.0
2. Desktop OS(windows linux or macOS):
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

xiaowenxia, 2019-03-14:
@xiake468 `out/xxx@mk3060/binary/` 目录下面有个readme.txt，有对固件相关说明。
mk3060增加了二级boot，需要注意一下：

- xxx@mk3060.ota.bin  是用来ota升级的固件。
- xxx@mk3060.bin 直接通过烧录工具烧录进去。

xiake468, 2019-03-14:
> @xiake468 out/xxx@mk3060/binary/ 目录下面有个readme.txt，有对固件相关说明。
> mk3060增加了二级boot，需要注意一下：
> 
> xxx@mk3060.ota.bin  是用来ota升级的固件。
> xxx@mk3060.bin 直接通过烧录工具烧录进去。
请问二级boot有说明是怎么用的吗,我用以前的方法(write 0x13200)烧写ota.bin的做法是不对的吗

xiake468, 2019-03-14:
> @xiake468 out/xxx@mk3060/binary/ 目录下面有个readme.txt，有对固件相关说明。
> mk3060增加了二级boot，需要注意一下：
> 
> xxx@mk3060.ota.bin  是用来ota升级的固件。
> xxx@mk3060.bin 直接通过烧录工具烧录进去。

mk3060 2.1.0的ota.bin的升级方式是不是改变了,我编译的example/blink都不能正常运行,我试了下在2.0.0下的ota.bin是没问题,现象就是程序不能运行

xiake468, 2019-03-15:
> @xiake468 out/xxx@mk3060/binary/ 目录下面有个readme.txt，有对固件相关说明。
> mk3060增加了二级boot，需要注意一下：
> 
> xxx@mk3060.ota.bin  是用来ota升级的固件。
> xxx@mk3060.bin 直接通过烧录工具烧录进去。

我注意到,2.1.0的更新提示中有如下:
* 提供统一的bootloader，实现适配后的控制器统一工具升级，并支持差分升级

另外board目录下的链接文件,链接的地址似乎也和以前不同了,现在看来可能是烧入方式改变了,但是具体怎么烧入还是不清楚呢

xiake468, 2019-03-15:
> @xiake468 out/xxx@mk3060/binary/ 目录下面有个readme.txt，有对固件相关说明。
> mk3060增加了二级boot，需要注意一下：
> 
> xxx@mk3060.ota.bin  是用来ota升级的固件。
> xxx@mk3060.bin 直接通过烧录工具烧录进去。
现在的流程应该是,先烧写二级boot,再根据二级boot的提示,烧写程序对吧? 我进入到2nd boot中,要求烧入,OS  partition, OTA partition,能否告诉我一下,这两个 partition分别对应哪个bin文件吗

xiake468, 2019-03-15:
> @xiake468 out/xxx@mk3060/binary/ 目录下面有个readme.txt，有对固件相关说明。
> mk3060增加了二级boot，需要注意一下：
> xxx@mk3060.ota.bin  是用来ota升级的固件。
> xxx@mk3060.bin 直接通过烧录工具烧录进去。
> 请问二级boot有说明是怎么用的吗,我用以前的方法(write 0x13200)烧写ota.bin的做法是不对的吗
对不起...我水平太差,我明白了...整个流程是先烧入2nd的boot,然后提示的两个是app 1,和ota的地址,第一次直接烧进app 1中就可以了,谢谢

xiake468, 2019-03-15:
> @xiake468 out/xxx@mk3060/binary/ 目录下面有个readme.txt，有对固件相关说明。
> mk3060增加了二级boot，需要注意一下：
> 
> xxx@mk3060.ota.bin  是用来ota升级的固件。
> xxx@mk3060.bin 直接通过烧录工具烧录进去。

对不起...我水平太差,我明白了...整个流程是先烧入2nd的boot,然后提示的两个是app 1,和ota的地址,第一次直接烧进app 1中就可以了,谢谢

closed 2019-03-15 by xiake468