alibaba/AliOS-Things #905: 用MQTTPacket_equals对比sha256摘要不合理

comporsa, 2019-03-15:
mqtt_client.c 第1372行的if (MQTTPacket_equals(&md5_topic, (char *)h->topic_filter)，其中后者h->topic_filter在WITH_MQTT_ZIP_TOPIC选项打开的情况下是SHA256摘要。
然而在这个函数里，使用strlen来获取这个摘要的长度，导致该函数完全不可能正常工作。

comporsa, 2019-03-15:
版本是2.1.0版

iotDevelop0319, 2019-03-19:
在MQTTPacket_equals函数中，只有当WITH_MQTT_ZIP_TOPIC没有定义的时候，才会去使用strlen，所以这个函数在我看来其实是没问题的。建议@comporsa 确认一下。

int MQTTPacket_equals(MQTTString *a, char *bptr)
{
    int alen = 0,
        blen = 0;
    char *aptr;
#if !(WITH_MQTT_ZIP_TOPIC)
    if (a->cstring) {
        aptr = a->cstring;
        alen = strlen(a->cstring);
    } else {
        aptr = a->lenstring.data;
        alen = a->lenstring.len;
    }
    blen = strlen(bptr);
#else
    aptr = a->lenstring.data;
    alen = a->lenstring.len;
    blen = alen;
#endif
    return (alen == blen) && (memcmp(aptr, bptr, alen) == 0);
}

closed 2019-03-25 by comporsa