alibaba/AliOS-Things #906: ESP8266一型一密动态注册成功但出现服务器无法解析的错误

simonliu009, 2019-03-15:
**Describe the bug**
首先在2.0.0，写死ProductKey/DeviceName/DeviceSecret的时候是没问题的。 
但是在2.0.0和2.1.0测试ESP8266一型一密动态注册的时候，可以获取DeviceSecret，但是都遇到了发送notify topic成功后无法解析服务器的错误
```
[022600]<I> topic:/sys/awss/event/connectap/notify
[022600]<D> payload:{"id":"4","version":"1.0","method":"awss.event.connectap.notify","params":{"awssVer":{"smartconfig":"2.0","zconfig":"2.0","router":"2.0","ap":"2.0","softap":"2.0"},"productKey":"a1gdHhOE5ji","deviceName":"c5fd3f9a95f74a144e52045a421f3887","mac":"60:01:94:16:63:9F","ip":"192.168.3.17","cipherType":3,"type":0}}

[022650]<I> send notify success
[022680]<I>  ok (0 skipped)
[022680]<I> Connecting to /a1gdHhOE5ji.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
mbedtls_net_connect 250: getaddrinfo fail- errno: 0
[022710]<E>  failed ! net_connect returned -0x0052
```

**environment**
1. branch(version): 
在rel_2.0.0和rel_2.1.0均遇到了了。但是rel2.0.0烧写了一次helloworld固件以后又烧回去就正常了，后来没有专门测试去复现。rel_2.1.0为了保持复现问题，一直反复烧写同一固件都不行。甚至擦除flash重新配网也不行。
固件是基于linkkit_example_solo.c修改的。

2. Desktop OS(windows linux or macOS):
macOS
3. example: 

**Additional context**
完整log太长，贴个文本附件
[getaddrinfo_fail_log.txt](https://github.com/alibaba/AliOS-Things/files/2971520/getaddrinfo_fail_log.txt)
。

xiaowenxia, 2019-03-18:
首先确定一下网络是否有问题，看看能不能ping通：`a1gdHhOE5ji.iot-as-mqtt.cn-shanghai.aliyuncs.com`

simonliu009, 2019-03-21:
更新一下：
1. 电脑ping随时都可以解析。
2. 换了手机热点可以解析。
3. 换回wifi又不行。
4. Wifi的情况下，我手动写了一个解析代码重复5秒执行，然后就连接就正常了。估计是sdk解析部分有问题？
目前暂时手动添加dns解析代码解决。

closed 2019-03-21 by simonliu009

simonliu009, 2019-03-21:
又测了几次，把定时器间隔减小到1秒，进入系统大约8秒左右才能正确解析，HAL接口运行前三次连接均失败，我的dns_gethostbyname（）运行到大约4-5次才能正确解析，此后HAL接口可以正确连接。

xiaowenxia, 2019-03-21:
你用esp8266连手机热点是正常解析的话，连你的wifi路由器解析太慢，估计和你的路由器或者运营商关系比较大。

获取域名使用的是esp8266的liblwip.a中的lwip_getaddrinfo()函数。

simonliu009, 2019-03-21:
wifi路由器解析太慢，解释不了默认HAL无论多少次都无法解析的问题。
运营商关系，解释不了同样wifi环境下，使用dns_gethostbyname()就能解析的问题。

JunJianX, 2020-12-07:
@simonliu009 请问您的问题解决了吗？

JunJianX, 2020-12-08:
@xiaowenxia 我也遇到了，请问有解决方法了吗？