alibaba/AliOS-Things #907: 新安装的win10   用的你们的win安装包。点对号一直报错。

wangshuai9517, 2019-03-16:
aos-cube version: 0.3.1
Check if required tools for stm32l432kc-nucleo exist
信息: 用提供的模式无法找到文件。
python.exe: can't open file './/build/scripts/aos_download_tools.py': [Errno 2] No such file or directory
make.exe: *** [build/build_rules/aos_kconfig.mk:87: .\build\kconfig\Win32\kconfig-conf.bat] Error 2
[AliOS-Things] ERROR: "C:/Users/swang/Desktop/Alios/AliosThing/rel_2.1.0/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "C:/Users/swang/Desktop/Alios/AliosThing/rel_2.1.0/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=C:/Users/swang/Desktop/Alios/AliosThing/rel_2.1.0/build -e -f build/Makefile helloworld@stm32l432kc-nucleo.config" in "C:\Users\swang\Desktop\Alios\AliosThing\rel_2.1.0"---
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

wangshuai9517, 2019-03-16:
vscode

xiaowenxia, 2019-03-18:
你是不是代码吗，没有全部下载下来？你看看`./build/scripts/aos_download_tools.py`这个文件存在不？

wangshuai9517, 2019-03-18:
在GitHub上都没有看见这个文件。我看了好几branch的源码里面都没看见这个文件。

xiaowenxia, 2019-03-18:
这个呀：https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/scripts/aos_download_tools.py

wangshuai9517, 2019-03-18:
> 你是不是代码吗，没有全部下载下来？你看看`./build/scripts/aos_download_tools.py`这个文件存在不？

尴尬了。两天前杠提交的。我那时候已经安装了好几天了。

wangshuai9517, 2019-03-18:
> 这个呀：https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/scripts/aos_download_tools.py

哭死了，我安装到半夜两点，然后就放弃了。估计第二天这个文件就添加进去了。

wangshuai9517, 2019-03-18:
谢谢，既然这个文件有了。就解决了。晚上我再去试试。非常感谢。

x709727942, 2019-04-15:
我也遇到这个问题了，你的解决了吗？

closed 2022-03-01 by skylarCai