alibaba/AliOS-Things #908: 用IDE编译第一个helloworld，结果就编译不通过，UnicodeDecodeError

gwfighter, 2019-03-18:
**Describe the bug**
按照wiki步骤编辑程序时出现错误：
“UnicodeDecodeError: 'utf8' codec can't decode byte 0xd6 in position 3: invalid continuation byte”。
找不到原因，配置正确。

**environment**
1. 最新版（刚下载） 
2. Windows 10
3. Helloworld

**Expected behavior**

**Screenshots**
> Executing task: aos make helloworld@aaboard_demo -c config <

aos-cube version: 0.3.1
Traceback (most recent call last):
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\aos\__main__.py", line 82, in main
    cli(None)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\aos\commands\make.py", line 33, in cli
    make_build(make_args)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\aos\commands\make.py", line 259, in make_build
    _run_make(['-e', '-f build/Makefile', make_args])
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\aos\commands\make.py", line 285, in _run_make
    make_cmd_str = ' '.join([make_cmd, 'HOST_OS=' + host_os, 'TOOLS_ROOT=' + tools_dir] + list(arg_list))
  File "c:\users\administrator\.aos\python-venv\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xd6 in position 3: invalid continuation byte
[AliOS-Things] ERROR: Unknown Error: 'utf8' codec can't decode byte 0xd6 in position 3: invalid continuation byte

**Additional context**

gwfighter, 2019-03-18:
搞了半天终于找到问题了，原来路径中不能有中文名字。。。这个坑对于初学者不知道该怎么避免。

xiaowenxia, 2019-03-18:
@gwfighter :+1: 谢谢你的反馈，后面我们优化一下

closed 2019-03-18 by gwfighter