alibaba/AliOS-Things #909: 运行alios things的时候wifi初始化异常

408056198, 2019-03-18:
**Describe the bug**

**environment**
1. AOS-R-2.0.0: 
2. windows:

**Expected behavior**
用庆科的EMW3165模块运行alios things的时候wifi初始化用了整整33s才进入application start，定位了一下是在mxchipInit里执行了33s,但是这个函数是个库函数，不知道里面执行了什么

**Screenshots**
[image]

[image]

**Additional context**

gwfighter, 2019-03-18:
估计这个要找庆科的人？
①庆科自己的Mico系统你试过没，里面好像也有WiFi例程。我以前玩过，貌似不需要这么长的初始化时间。

408056198, 2019-03-19:
> 估计这个要找庆科的人？
> ①庆科自己的Mico系统你试过没，里面好像也有WiFi例程。我以前玩过，貌似不需要这么长的初始化时间。

@gwfighter 是庆科的人让我在这里留言找解决方法的，mico系统我用过的，那里的wifi初始化确实很快，但是我现在想连接阿里云，直接用mico不是很方便，所以想跑alios things的，我试过好几块EMW3165的板子，都需要33s，你知道alios things例程序里的庆科模块的芯片驱动是哪边移植的吗？庆科还是阿里？

gwfighter, 2019-03-19:
> > 估计这个要找庆科的人？
> > ①庆科自己的Mico系统你试过没，里面好像也有WiFi例程。我以前玩过，貌似不需要这么长的初始化时间。
> 
> @gwfighter 是庆科的人让我在这里留言找解决方法的，mico系统我用过的，那里的wifi初始化确实很快，但是我现在想连接阿里云，直接用mico不是很方便，所以想跑alios things的，我试过好几块EMW3165的板子，都需要33s，你知道alios things例程序里的庆科模块的芯片驱动是哪边移植的吗？庆科还是阿里？
不是吧。。。我还没走到那一步。我也是在测试用庆科的板子连接阿里云，mk3060你试过没？alios 用这个板子做demo讲解编译环境的（https://github.com/alibaba/AliOS-Things/wiki/MK3060-Hardware-Setup）

gwfighter, 2019-03-19:
我感觉这个应该是庆科，阿里做好，测试好，然后贡献到代码库。

gwfighter, 2019-03-19:
我这两天也在尝试在3165上运行alios，download还没有成功。你这边有经验记录么？我感觉资料好乱，得一点点摸索。好累。

408056198, 2019-03-20:
@gwfighter 你编译过了吗

408056198, 2019-03-20:
@gwfighter 你邮箱多少？可以一起探讨一下

gwfighter, 2019-03-20:
可以的，一起学习打怪升级。o(*￣︶￣*)o。我的邮箱是：ygmchina@qq.com，你的邮箱呢？

closed 2022-03-01 by skylarCai