alibaba/AliOS-Things #917: Linux 64

DorBlack, 2019-03-19:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Additional context**

DorBlack, 2019-03-19:
[AliOS-Things] ERROR: "/home/unke/AliOS-Things/build/cmd/linux64/make" returned error code 2.
[AliOS-Things] ERROR: Command "/home/unke/AliOS-Things/build/cmd/linux64/make HOST_OS=Linux64 TOOLS_ROOT=/home/unke/AliOS-Things/build -e -f build/Makefile helloworld@linuxhost" in "/home/unke/AliOS-Things"
make 无效

xiaowenxia, 2019-03-20:
`aos make VERBOSE=1`看看详细的信息输出。

DorBlack, 2019-03-23:
aos-cube version: 0.3.1
Please configure first:
*********************************************************************
$ aos make <app>@<board> -c config
---> Or more config: $ aos make menuconfig
*********************************************************************
build/Makefile:52: *** No Config Found!。 停止。
[AliOS-Things] ERROR: "/home/unke/AliOS-Things/build/cmd/linux64/make" returned error code 2.
[AliOS-Things] ERROR: Command "/home/unke/AliOS-Things/build/cmd/linux64/make HOST_OS=Linux64 TOOLS_ROOT=/home/unke/AliOS-Things/build -e -f build/Makefile VERBOSE=1" in "/home/unke/AliOS-Things"
---
这个就是完整的信息

xiaowenxia, 2019-03-24:
你输入的命令是啥呢？aliosthings 2.1的命令跟之前的版本有变化。

DorBlack, 2019-03-24:
aos make helloworld@stm32l476rg-nucleo

xiaowenxia, 2019-03-25:
@DorBlack 新的命令发生变化了哈，参考[文档](https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-uCube.zh)。

walker2048, 2019-03-27:
国内有个交流工具叫钉钉，里面大佬挺多。

walker2048, 2019-04-24:
@DorBlack 请问你的Linux环境是什么发行版？
我这边用Alpine Linux就会出现这个错误，目前排查应该是与Alpine Linux使用的不是标准glibc库有关系

closed 2022-03-01 by skylarCai