alibaba/AliOS-Things #924: developerkit UART 0 OVER RUN

kentzh, 2019-03-23:
在developerkit板子上，使用rel_2.1.0版本编译并下载otaapp程序后，在命令行输入OTA测试命令：
[
OTA_APP "a154F8BTLOS" "test01" "ojjE88iqT2dilHgamHv0Zu4TKwCXcU38" "TVekLkQ7Es0frNNY"
]

会概率性出现“FATAL UART 0 OVER RUN ”错误，错误发生概率在80%以上

-------------------LOG INFO---------------
[2019-03-23 10:30:47.416]# RECV ASCII>
boot up ...
aos Bank2 bootup!

[2019-03-23 10:30:47.507]# RECV ASCII>
uart 1 enter uart_receive_start_dma instance 0x40004800[000050]<V> sensor:  drv_acc_st_lsm6dsl_init successfully 

[000060]<V> sensor:  drv_gyro_st_lsm6dsl_init gyro do not need reset

[000070]<V> sensor:  drv_gyro_st_lsm6dsl_init successfully 

[000080]<V> sensor:  drv_als_liteon_ltr553_init successfully 

[000090]<V> sensor:  drv_ps_liteon_ltr553_init successfully 

[000100]<V> sensor:  drv_baro_bosch_bmp280_init successfully 

[000120]<V> sensor:  drv_mag_memsic_mmc3680kj_init successfu

[2019-03-23 10:30:47.582]# RECV ASCII>
lly 

[000130]<V> sensor:  drv_humi_sensirion_shtc1_init successfully 

[000130]<V> sensor:  drv_temp_sensirion_shtc1_init successfully 

             Welcome to AliOS Things           

[2019-03-23 10:30:49.642]# RECV ASCII>
[002150]<I> wifi bk7231 FW version is: FWVER:APP-1.1.6-20180928.1547

OK

[002170]<I> Will connect via at cmd: AT+WJAP=CMCC-whxf,h9djj6ef

[002200]<I> AT command AT+WJAP=CMCC-whxf,h9d986ef succeed, rsp: 
OK

[2019-03-23 10:30:55.137]# RECV ASCII>
[007630]<I> AT command AT+WMAC? succeed, rsp: 
+WMAC:e47deb1a5f86

OK

[007640]<I> mac result: e47deb1a5f86

[007640]<I> Got ip : 192.168.1.2, gw : 192.168.1.1, mask : 255.255.255.0

[007670]<I> AT command AT+WMAC? succeed, rsp: 
+WMAC:e47deb1a5f86

OK

[007680]<I> mac result: e47deb1a5f86

[007680]<V> Hello, WiFi GOT_IP event! at board/developerkit/aos/board_cli.c 210

[007690]<V> Hello, WiFi GOT_IP event! at board/developerkit/aos/board_cli.c 210

[2019-03-23 10:30:58.728]# SEND ASCII>
OTA_APP "a154F8BTLOS" "test01" "ojjE88iqT2dilHgamHv0Zu4TKwCXcU38" "TVekLkQ7Es0frNNY"

[2019-03-23 10:30:58.803]# RECV ASCII>
OTA_APP "a154F8BTLOS" "test01" "ojjE88iq### FATAL UART 0 OVER RUN !!!
# ### FATAL UART 0 OVER RUN !!!
# lHgamHv0Zu4TKwCXcU38" "TVekLkQ7Es0frNNY"
#

Jianlong0430, 2019-03-25:
@kentzh Please update Wi-Fi firmware to 201812 instead of APP-1.1.6-20180928.1547(from above log), the guide is available on github wiki.

Jianlong0430, 2019-03-25:
@kentzh please show whole logs to us. please input cli again if log shows '"# ## FATAL UART OVER RUN !!!". It is normal case for '# ## FATAL UART OVER RUN !!!', but abnormal for 80% failure.

kentzh, 2019-03-28:
After WIFI FW updated, this issue disappeared. Thanks!

closed 2019-03-28 by kentzh