alibaba/AliOS-Things #925: Build helloworld@starterkit failed

Andy-J20, 2019-03-23:
**Describe the bug**
Build helloworld@starterkit 时，显示 The terminal process terminated with exit code: 2， 无法Build
**environment**
1. branch(version): alios-studio 0.11.2
2. Desktop OS(windows linux or macOS):macOS
3. example: Build时，显示错误代码

**Expected behavior**

**Screenshots**

[image]

**Additional context**

comporsa, 2019-03-23:
aos 最新版是0.3，升级下aos再重新编译

Andy-LZH, 2019-03-24:
感谢comporsa，xiaowen, wenzong等大牛讲解。问题如comorsa所说，为aos-cube版本问题。须更新到，0.3以上版本。而我的问题是，在terminal上更新完成后，且 `aos --version` 后依旧无法编译，报同样的错。后发现，aos-cube同时存在于 `Library > Framework` 以及 `Users > .aos`中。使用terminal首先卸载aos-cube，手动删除 `Users > .aos`  目录，最后卸载VS code中的python（全局环境下已安装python）后重新在VS code中alios-studio安装aos-cube后，问题解决。

closed 2019-03-24 by Andy-LZH