alibaba/AliOS-Things #927: 编译提示错误

kookeda, 2019-03-24:
**Describe the bug**
使用  aos make helloworld@esp8266  编译出现错误，另外无法切换编译目标。
后使用  aos make uart@esp8266 提示错误后，使用aos make 仍然在编译helloworld工程。

**environment**
1. AliOS-Things (rel_2.1.0)
2. windows7 旗舰版

**Expected behavior**
Administrator@Admin-PC MINGW64 /d/aos/AliOS-Things (rel_2.1.0)
$ aos make uart@esp8266
aos-cube version: 0.3.2
Check if required tools for esp8266 exist

Config completed, please run build directly: $ aos make

build/Makefile:129: *** Build Target Deprecated!.  Stop.
[AliOS-Things] ERROR: "D:/aos/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "D:/aos/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=D:/aos/AliOS-Things/build -e -f build/Makefile uart@esp8266" in "D:\aos\AliOS-Things"
---

Administrator@Admin-PC MINGW64 /d/aos/AliOS-Things (rel_2.1.0)
$ aos make
aos-cube version: 0.3.2
Check if required tools for esp8266 exist
ECHO 处于关闭状态。
Build Configuration:
====================
App:   helloworld
Board: esp8266
====================
ECHO 处于关闭状态。
Build AOS Now
TOOLCHAIN_PATH=
信息: 用提供的模式无法找到文件。
Making helloworld@esp8266.bin

                        AOS MEMORY MAP
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| arch_xtensa_lx106                        | 4066      | 2784     |
| board_esp8266                            | 1322      | 0        |
| debug                                    | 3896      | 16       |
| helloworld                               | 100       | 0        |
| kernel_init                              | 117       | 0        |
| kv                                       | 2244      | 24       |
| libcirom                                 | 27245     | 1200     |
| libcrypto                                | 14506     | 0        |
| libgcc                                   | 2190      | 0        |
| libhal                                   | 37        | 0        |
| liblwip                                  | 29137     | 1595     |
| libmain                                  | 29466     | 708      |
| libnet80211                              | 42566     | 3209     |
| libphy                                   | 48154     | 461      |
| libpp                                    | 44409     | 18005    |
| libwpa                                   | 17236     | 626      |
| mcu_esp8266                              | 9706      | 576      |
| netmgr                                   | 80        | 8        |
| newlib_stub                              | 367       | 0        |
| osal_aos                                 | 765       | 0        |
| ota_2nd_boot                             | 18848     | 40       |
| rhino                                    | 11692     | 3228     |
| ulog                                     | 318       | 6        |
| vfs                                      | 1337      | 1105     |
| yloop                                    | 1904      | 32       |
| *fill*                                   | 1545      | 105      |
|=================================================================|
| TOTAL (bytes)                            | 313253    | 33728    |
|=================================================================|
bin crc: 401638b1
315588
4d6c0ace4e80e56f346f1df6554720d7
Build complete: helloworld@esp8266

**Screenshots**

**Additional context**

simonliu009, 2019-03-24:
2.1.0编译方式变了
需要 `aos make menuconfig` 进行编译选项设置
或者`aos make helloworld@esp8266 -c config` 创建默认配置文件
然后
`aos make` 编译
`aos upload`上传

kookeda, 2019-03-24:
编译成功后，烧录如硬件设备，没有正常输出，会是哪里原因呢？

simonliu009, 2019-03-24:
AliOS Things入门可以参考我写的教程：
[SimonLiu的ESP8266与AliOS Things 学习教程系列目录 - SimonLiu的博客 - CSDN博客](https://blog.csdn.net/toopoo/article/details/86920755)

kookeda, 2019-03-24:
感谢 @simonliu009  热心解答。我这个的原因终于弄清楚了，我的模块SPI  FLASH使用的是DOUT模式,所以使用QIO下载不能正常工作，报flash read err。修改一下，upload配置的json文件，顺利可以使用vscode 下载了，暂时还没搞明白，怎么动态设置SPI  mode，后面有时间再研究。

closed 2019-03-24 by kookeda