alibaba/AliOS-Things #928: ESP8266 OTA 失败，提示 upgrade over err:-17

simonliu009, 2019-03-24:
**Describe the bug**
1. 基于linkkitapp修改的代码，测试OTA的时候报错，然后反复重启尝试OTA仍然到同一个地方提示此错误，重新上电也是如此。
 错误内容：upgrade over err:-17
2. 固件验证进度一直是0.
3. 关闭固件验证进度弹窗之后，居然现实验证成功。

**environment**
1. branch(version): 
master branch
2. Desktop OS(windows linux or macOS):
macOS
3. example: 

**Expected behavior**

**Screenshots**
[image]
[image]

**Additional context**
Log：
[008270]<I> ....................................................
[008270]<I>           ProductKey : a1gdHhOE5ji
[008270]<I>           DeviceName : c5fd3f9a95f74a144e52045a421f3887
[008280]<I>             DeviceID : a1gdHhOE5ji.c5fd3f9a95f74a144e52045a421f3887
[008290]<I> ....................................................
[008300]<I>        PartnerID Buf :
[008300]<I>         ModuleID Buf :
[008310]<I>           Guider URL :
[008310]<I>       Guider SecMode : 2 (TLS + Direct)
[008320]<I>     Guider Timestamp : 2524608000000
[008320]<I> ....................................................
[008340]<I> -----------------------------------------
[008340]<I>             Host : a1gdHhOE5ji.iot-as-mqtt.cn-shanghai.aliyuncs.com
[008350]<I>             Port : 1883
[008360]<I>         ClientID : a1gdHhOE5ji.c5fd3f9a95f74a144e52045a421f3887|securemode=2,timestamp=2524608000000,signmethod=hmacsha256,gw=0,ext=0,a=aos-r-2.1.0|
[008380]<I>       TLS PubKey : 0x4020ca3f ('-----BEGIN CERTI ...')
[008390]<I> -----------------------------------------
[008400]<I> MQTT init success!
[008400]<I> Loading the CA root certificate ...
[008440]<I>  ok (0 skipped)
[008440]<I> Connecting to /a1gdHhOE5ji.iot-as-mqtt.cn-shanghai.aliyuncs.com/1883...
[008460]<I>  ok
[008460]<I>   . Setting up the SSL/TLS structure...
[008460]<I>  ok
[008460]<I> Performing the SSL/TLS handshake...
[009430]<I>  ok
[009440]<I>   . Verifying peer X.509 certificate..
[009440]<I> certificate verification result: 0x00
[009500]<I> mqtt connect success!
[009500]<E> the network interface info set failed or not set, writen len is 0
[009520]<I> Upstream Topic: '/sys/a1gdHhOE5ji/c5fd3f9a95f74a144e52045a421f3887/thing/deviceinfo/update'
[009520]<I> Upstream Payload:
␛[1;33m
> {
>     "id": "0",
>     "version": "1.0",
>     "params": [
>         {
>             "attrKey": "SYS_ALIOS_ACTIVATION",
>             "attrValue": "0201000088199BBC60019416639F010100050009PLAIN",
>             "domain": "SYSTEM"
>         },
>         {
>             "attrKey": "SYS_LP_SDK_VERSION",
>             "attrValue": "2.3.0",
>             "domain": "SYSTEM"
>         },
>         {
>             "attrKey": "SYS_SDK_LANGUAGE",
>             "attrValue": "C",
>             "domain": "SYSTEM"
>         },
>         {
>             "attrKey": "SYS_SDK_IF_INFO",
>             "attrValue": "invalid network interface info",
>             "domain": "SYSTEM"
>         }
>     ],
>     "method": "thing.deviceinfo.update"
> }

␛[0m[009630]<I> Upstream Topic: '/ota/device/inform/a1gdHhOE5ji/c5fd3f9a95f74a144e52045a421f3887'
[009640]<I> Upstream Payload:
␛[1;33m
> {
>     "id": "1",
>     "params": {
>         "version": "app-1.0.0-20190323.2139"
>     }
> }

␛[0m[009660]<V> MQTT Construct  OTA start
[009660]<I> Public name:/ota/device/inform/a1gdHhOE5ji/c5fd3f9a95f74a144e52045a421f3887 msg:{"id":0,"params":{"version":"app-1.0.0-20190323.2139"}}
[009690]<I> Upstream Topic: '/ota/device/inform/a1gdHhOE5ji/c5fd3f9a95f74a144e52045a421f3887'
[009690]<I> Upstream Payload:
␛[1;33m
> {
>     "id": 0,
>     "params": {
>         "version": "app-1.0.0-20190323.2139"
>     }
> }

␛[0m[009710]<I> upgrade:/ota/device/upgrade/a1gdHhOE5ji/c5fd3f9a95f74a144e52045a421f3887
[009730]<I> mqtt subscribe packet sent,topic = /ota/device/upgrade/a1gdHhOE5ji/c5fd3f9a95f74a144e52045a421f3887!
[009760]<I> Downstream Topic: '/ota/device/upgrade/a1gdHhOE5ji/c5fd3f9a95f74a144e52045a421f3887'
[009760]<I> Downstream Payload:
␛[1;33m
< {
<     "code": "1000",
<     "data": {
<         "size": 590336,
<         "sign": "a51c363173b54d3a3bdc8cf3d4f952ac",
<         "version": "app-1.0.0-20190324.1050",
<         "url": "https: //iotx-ota.oss-cn-shanghai.aliyuncs.com/CCRpJ38TKD_1553395903392.bin?Expires=1553485285&OSSAccessKeyId=cS8uRRy54RszYWna&Signature=o1j5QIunF7LdOyKwI8bCIu%2FzrCs%3D",
<         "signMethod": "Md5",
<         "md5": "a51c363173b54d3a3bdc8cf3d4f952ac"
<     },
<     "id": 1553398885865,
<     "message": "success"
< }

␛[0m[009830]<I> mqtt cb evt:12 {"code":"1000","data":{"size":590336,"sign":"a51c363173b54d3a3bdc8cf3d4f952ac","version":"app-1.0.0-20190324.1050","url":"https://iotx-ota.oss-cn-shanghai.aliyuncs.com/CCRpJ38TKD_1553395903392.bin?Expires=1553485285&OSSAccessKeyId=cS8uRRy54RszYWna&Signature=o1j5QIunF7LdOyKwI8bCIu%2FzrCs%3D","signMethod":"Md5","md5":"a51c363173b54d3a3bdc8cf3d4f952ac"},"id":1553398885865,"message":"success"}
[010040]macStr= 60019416639f
[010040]Device Name: c5fd3f9a95f74a144e52045a421f3887
[010040]<V> ============free heap size =25568==========
[010890]<E> upgrade over err:-17
reboot!
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/9706670
del if0
usl
␂@�Z-nP␙␕␓�M␌%��␇       %Y␎+Q
        �E␋x��R␅␄␄␓␑␑   )dP��)5Y
␓␅
L�C�    I�V␑�-��T␑��␆␑  )l�␝    5Y
␑␋�NPA␌␋�К�P9j)�␀␁��C␄␁��A�␀�I*␑R�e��l�␕%␀�Q@5␋�NRA� �ڻ^␑�)��C␄����xx���rZ␜

xiaowenxia, 2019-03-25:
@simonliu009 能不能抓一个完整的log看看哈。

xiaowenxia, 2019-03-25:
@simonliu009 你的esp8266 flash大小是？

simonliu009, 2019-03-25:
@xiaowenxia 
我的是ESP-12F, 32Mbit的大小。
完整log请参看附件。
为了避免信息混淆在一起，我是在启动40秒后收到MQTT ping的回复（第407行：[040020]<I> receive ping response!)后开始推送。
然后经过了upgrade over err:-17 ，自动重启，然后服务器自动再推送，再次upgrade over err:-17，然后自动重启（第二次重启后我就停止log了，后面都是一样的循环失败重启）。
[upgrade over err log.txt](https://github.com/alibaba/AliOS-Things/files/3003123/upgrade.over.err.log.txt)

xiaowenxia, 2019-03-26:
@simonliu009 你有更改过哪些地方吗？看起来可能是flash map划分得不正确。

simonliu009, 2019-03-26:
@xiaowenxia 
flash map我没做改动，都是默认aos upload烧写的。

simonliu009, 2019-03-26:
在ali的帮助下问题已解决。
OTA问题是由于`flash map`引起，使用`system_get_flash_size_map()`打印flash map发现为4。然后我把`build/site_scons/upload/esp8266.json`的烧写选项从 "--flash_size", "detect", 改为 "--flash_size", "4MB-c1"。flash map就是6了。但是默认编译选项在`platform/mcu/esp8266/gen_crc_bin.mk`中的` python ../../../platform/mcu/esp8266/tools/gen_appbin.py $(CLEANED_BUILD_STRING)$(RADIXPOINT)$(BINSTYPE_LOWER)$(LINK_OUTPUT_SUFFIX) 2 1 15 5 `最后一个参数,参数是5。
根据 `platform/mcu/esp8266/tools/gen_appbin.py` 的定义:
   #-------------------
    #flash_size_map=
    #     0 : 512 KB (256 KB + 256 KB)
    #     1 : 256 KB
    #     2 : 1024 KB (512 KB + 512 KB)
    #     3 : 2048 KB (512 KB + 512 KB)
    #     4 : 4096 KB (512 KB + 512 KB)
    #     5 : 2048 KB (1024 KB + 1024 KB)
    #     6 : 4096 KB (1024 KB + 1024 KB)
    #-------------------
目前是5 : 2048 KB (1024 KB + 1024 KB)
我的ESP-12F应该改为6。不过目前5也能正常运行和OTA推送。比较奇怪，可能可以兼容？需要进一步测试。

closed 2019-03-26 by simonliu009