alibaba/AliOS-Things #929: Windows 下vscode烧写问题

DorBlack, 2019-03-25:
> Executing task: aos upload helloworld@stm32l476rg-nucleo <

aos-cube version: 0.3.2
[INFO]: Currently in aos_sdk_path: 'F:\STM32\AliOS-Things'
[INFO]: Target: helloworld@stm32l476rg-nucleo

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "build\site_scons\site_init.py", line 39, in aos_global_config
scons: Reading SConscript files ...
[INFO]: Scons to upload!args:{'APPLICATION': 'helloworld', 'COMMAND': 'upload', 'BOARD': 'stm32l476rg-nucleo'}

[INFO]: Target: helloworld@stm32l476rg-nucleo
[INFO]: Currently in aos_sdk_path: 'F:\STM32\AliOS-Things'

[INFO]: Running cmd:
        'F:\STM32\AliOS-Things/build/cmd/win32/st-flash.exe --reset write F:\STM32\AliOS-Things/out/helloworld@stm32l476rg-nucleo/binary/helloworld@stm32l476rg-nucleo.bin
0x08000000'
st-flash 1.5.0
2019-03-25T22:28:07 INFO common.c: Loading device parameters....
2019-03-25T22:28:07 INFO common.c: Device connected is: L4 device, id 0x10076415
2019-03-25T22:28:07 INFO common.c: SRAM size: 0x18000 bytes (96 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
2019-03-25T22:28:07 INFO common.c: Ignoring 4 bytes of 0xff at end of file
2019-03-25T22:28:07 INFO common.c: Attempting to write 59112 (0xe6e8) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0800e000 erasedEraseFlash - Page:0x1c Size:0x800
2019-03-25T22:28:08 INFO common.c: Finished erasing 29 pages of 2048 (0x800) bytes
2019-03-25T22:28:08 INFO common.c: Starting Flash write for F2/F4/L4
2019-03-25T22:28:08 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_GETSTATUS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_GETSTATUS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_GETSTATUS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_GETSTATUS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_GETSTATUS

xiaowenxia, 2019-03-26:
@DorBlack seems ST-Link is not working, this problem always appears when `flash`? try reinstall st-link driver.

DorBlack, 2019-03-26:
重新安装了驱动，还是不行，烧写未完成，字节数对不上。我用了keil，在keil上能够成功。

walker2048, 2019-03-27:
建议插上板子，截图下驱动。有些系统装的驱动不对，只有部分工具能烧录。或者你试试STlink util包里带的GUI烧录工具看看能否连上。