alibaba/AliOS-Things #931: esp8266 node mcu烧写和运行问题

legenddcr, 2019-03-27:
**Describe the bug**
esp8266放在board目录中，实际上这是一个SoC platform，这种组织并不合理。导致目前AOS的代码库实际上不能对node mcu开发板进行AliOS Things的烧写，需要手动修改烧写命令行，增加flash mode参数，并且代码库中esp8266的烧写波特率921600，node mcu v3使用的串口芯片CH340G无法支持。

**environment**
1. branch(version): 
All branches.

2. Desktop OS(windows linux or macOS):
Not related with host OS

3. example: 

**Expected behavior**
1、建议将esp8266烧写波特率调整为115200，能够同时支持node mcu v2和v3两种开发板的烧写。
2、目前烧写的boot镜像配置了921600的波特率，在node_mcu v3上烧写后串口日志也无法正常运行，建议给出解决方案，或者调整默认的boot镜像波特率

**Screenshots**

**Additional context**

walker2048, 2019-03-27:
board目录指的是板级代码结构，而不是平台代码（平台代码在platform/mcu/esp8266目录）。我觉得没什么不合理的，请再看一下代码，另外烧写波特率和串口交互波特率都可以自行调整。

walker2048, 2019-03-27:
的确，我刚开始接触alios-things的时候，也觉得代码结构有毛病，platform也放，board目录也放。后来仔细研究代码后觉得，这样的组织结构确实更合理，平台级的通用代码放在platform目录。根据不同板子，设置不同的board代码来定义不同的Pin脚，板上资源等等内容。哪怕像esp32，esp8266这样的wifi模块，也有可能根据不同项目需求，定制了不同的硬件底板，最终导致板上资源或者可复用的IO定义不能使用同样的标准。

legenddcr, 2019-03-27:
> board目录指的是板级代码结构，而不是平台代码（平台代码在platform/mcu/esp8266目录）。我觉得没什么不合理的，请再看一下代码，另外烧写波特率和串口交互波特率都可以自行调整。

参数可以自己通过代码调整，不过缺少的应该是对应的说明或文档，对不同的市面上常见的esp8266开发板应该告诉用户需要调整哪些参数、怎么调整，要不然需要用户自行摸索解决。

board目录下放esp8266目录这么做合理的话，那为什么对于esp32也不做类似处理让用户自行调整，而要在board目录里放上具体esp32devkitc的开发板？

walker2048, 2019-03-27:
源码既是文档，8266的RAM太小，建议早点跳坑。除掉wifi模块占用的50K左右内存，可用内存才40多K。目前了解情况的人都脱坑了，8266代码更新自然就少，更不用说还针对8266不同板子去适配。

walker2048, 2019-03-27:
alios-things的文档确实不是很全面。但是作为兼容大量平台的系统来说，我觉得已经不错了。Linux发展那么多年了，也不是每个arm设备都能适配的。天涯何处无芳草，何必单恋一枝花。能运行alios-things的wifi模组多了去了。另外在文档方面，短时间内是比较难出具特别全面的文档的，尤其是针对某块板子的文档。毕竟这个系统不是针对某块板子开发的系统。

imraax, 2019-03-27:
>参数可以自己通过代码调整，不过缺少的应该是对应的说明或文档，对不同的市面上常见的esp8266开发板应该告诉用户需要调整哪些参数、怎么调整，要不然需要用户自行摸索解决。

我也曾反映过这个问题，得到的回复是自行摸索...
我看example中好像可以通过uart重新设置速率，就翻到hal里的uart也看到有相关结构体定义，然后也看了platform中的参数。我还是很迷，貌似两种方法都可行。关于这个问题我跟walker2048也探讨过好几天，最终不了了之。
文档不全，这对新人的确是不太友好的，我也建议官方完善一下相关文档。

taksing-leung, 2019-03-27:
本人提供有偿指导，qq123004304

zlasd, 2019-05-14:
提供一份可用的烧录配置`build/site_scons/upload/esp8266.json`
``` 
{
    "cmd": [
        {
            "Linux32": "esptool.py",
            "Linux64": "esptool.py",
            "OSX": "esptool.py",
            "Win32": "esptool.exe"
        },
        "--chip",
        "esp8266",
        "--port",
        "@PORT@",
        "--baud",
        "115200",
        "write_flash",
        "--flash_mode",
        "dout",
        "--flash_size",
        "4MB", 
        "0x0",
        "@AOSROOT@/platform/mcu/esp8266/bsp/boot_v1.7_921600.bin",
        "0x3fc000",
        "@AOSROOT@/platform/mcu/esp8266/bsp/esp_init_data_default.bin",
        "0x3fe000",
        "@AOSROOT@/platform/mcu/esp8266/bsp/blank.bin",
        "0x1000",
        "@AOSROOT@/out/@TARGET@/binary/@TARGET@.bin"
    ]
}
```

closed 2022-03-01 by skylarCai