alibaba/AliOS-Things #935: 针对开发板的差分升级功能

kentzh, 2019-03-28:
developerkit开发板，希望验证差分升级功能，以方便后期连接NB模块时使用，但发现云端似乎不支持该开发板，能否还有其它方法进行验证？

另外，后期云端有开通搭载了AliosThings的其它设备的差分升级功能吗？

xiaowenxia, 2019-03-29:
@kentzh 请问你是哪个NB模块呢？

kentzh, 2019-03-29:
现在还没定，应该会用BC26或SIM7020，NB的带宽有限，希望能用到差分升级的功能

xiaowenxia, 2019-03-29:
NB模块目前还不支持差分升级哈。

kentzh, 2019-03-29:
依据https://help.aliyun.com/document_detail/58328.html?spm=5176.11485173.0.0.7e7859afV2DD7b的介绍，差分升级采用了切片的方式，这样可以减少更新数据量，节约时间和流量，应该和使用什么模块没有关系吧？
我们采用的是MCU+NB模块的方式，AliosThings跑在MCU上，NB只是通路啊

kentzh, 2019-03-29:
在developerkit开发板上，使用WIFI已经做过OTA验证，整包OTA是可以的，现在想验证下差分OTA

xiaowenxia, 2019-04-01:
> 依据[https://help.aliyun.com/document_detail/58328.html?spm=5176.11485173.0.0.7e7859afV2DD7b的介绍，差分升级采用了切片的方式，这样可以减少更新数据量，节约时间和流量，应该和使用什么模块没有关系吧？](https://help.aliyun.com/document_detail/58328.html?spm=5 [...]

那你是用的什么MCU？目前差分升级支持的MCU有限，你也可以自己适配差分升级功能哈。

taotaodong527, 2019-10-10:
Are there any strictions to support OTA Diff feature? Such as RAM and Flash requirements? Are there any specific docunments of how to porting OTA Diff feature

closed 2022-03-01 by skylarCai