alibaba/AliOS-Things #943: linkkit_sample_solo例程接收成功后处理函数post_property_cb没有被调用

jxjxjinxin, 2019-04-08:
linkkit_sample_solo例程中属性消息发送后的回调函数post_property_cb没有被调用，这个是自己的配置没有配好还是这个机制本身就还没有完善好？

fenghr1219, 2019-04-08:
@jxjxjinxin 您这里是使用的什么版本的alios-things？

fenghr1219, 2019-04-08:
@jxjxjinxin 您这里可以提供设备端的log 然后我们分析一下看看

Bloodance, 2019-04-08:
有收到post_reply消息吗？

jxjxjinxin, 2019-04-08:
> @jxjxjinxin 您这里可以提供设备端的log 然后我们分析一下看看

[image]
这个主要是我移植到mico平台了，所以我也不知道是不是alios-thing的原因...

jxjxjinxin, 2019-04-08:
> 有收到post_reply消息吗？

没有收到post_reply消息

fenghr1219, 2019-04-08:
@jxjxjinxin 您这里不要使用master分支的代码 您这里先使用rel分支的代码试试？

jxjxjinxin, 2019-04-08:
> @jxjxjinxin 您这里不要使用master分支的代码 您这里先使用rel分支的代码试试？

我仔细调试了下，看了下我自己订阅topic里面也没有订阅post_reply，应该是被我改成透传的topic了，这个透传和Alink协议使用哪个在SDK里面是不是有宏定义，这个在哪里的，应该是被我改了0.0

jxjxjinxin, 2019-04-08:
问题解决了，是我没有订阅相应的reply消息， 谢谢拉！

fenghr1219, 2019-04-08:
@jxjxjinxin 好的 感谢您的支持