alibaba/AliOS-Things #944: 内存问题

xiaoyang432, 2019-04-08:
**Describe the bug**
请问为什么新的branch会多消耗70多KB的内存 ？

**environment**
1. branch(version): rel_2.1.0
2. Desktop OS(windows linux or macOS):windows
3. example: linkkitapp

**Expected behavior**

**Screenshots**

**Additional context**

xiaowenxia, 2019-04-08:
@xiaoyang432 应该不会消耗这么多内存的，有可能mcu的**栈**更改过，或者一些全局静态变量更改为了**malloc**这样的动态内存。

xiaoyang432, 2019-04-09:
整个SDK都没有改动过，直接编译烧录就发现多耗了70多KB的内存，该问题可以直接复现的，我测试的MCU是esp32devkitc

xiaoyang432, 2019-04-10:
@xiaowenxia 找到具体的问题点了，是AliOS-Things\platform\mcu\esp32\bsp\entry.c 中 hal_wifi_init();的问题，如果运行该函数就会消耗70多KB的内存，之前的版本没有这个问题的，请帮忙解决，谢谢！

xiaowenxia, 2019-04-10:
@xiaoyang432 感谢你反馈的问题哈:smile:，你的统计方法能贴一下哈，我们内部排查下。

xiaoyang432, 2019-04-11:
@xiaowenxia 对，主要是esp_wifi_init(&cfg)这个函数消耗了40多KB的内存，统计方法及窗口打印的数据如截图所示
[image]
[image]

closed 2022-03-01 by skylarCai