alibaba/AliOS-Things #946: ESP8266 Uart0只能接收26Byte串口数据

Jiangwei0802, 2019-04-11:
**Describe the bug**

**environment**
1. branch(version): 
aos-2.1.0
2. Desktop OS(windows linux or macOS):
windows
3. example: 
通过串口助手给uart0发送31个字节数据，运行app/example/uart例程；接收到的数据低于26个字节的均接收正常，超过26的就只能接收26个字节
**Expected behavior**
运行乐鑫官网sdk，测试串口接收正常；

**Screenshots**

[image]

**Additional context**

Detao, 2019-04-15:
我也遇到这个问题 
我把接收队列buffer改大就可以了
修改了hal/uart.c
`#define UART0_BUF_QUEUE_BYTES 1024`
以及bsp/driver/uart.c 中的
` uint8 fifo_tmp[1024] = {0};`
原本都是128

[image]

Jiangwei0802, 2019-04-20:
奇怪的，我改成1280也不行，改为1024就正常！

closed 2019-04-20 by Jiangwei0802