alibaba/AliOS-Things #950: emw3060 debug日志打印问题

Torchwoods, 2019-04-15:
**Describe the bug**
在ubuntu下使用rel2.1.0分子编译emw3060的helloworld工程，串口调试日志没有输出。采用默认的工程，没有改动配置，串口波特率为912600，我使用rel2.0.0没有什么问题，可以正常输出内容，但是使用rel2.1.0日志没有输出。

**environment**
1. rel2.1.0
2. ubuntu 16
3. helloworld

xiaowenxia, 2019-04-16:
@Torchwoods rel2.1.0的helloworld@mk3060能正常运行的，你烧录的是哪个binary呢？正常的是`out/helloworld@mk3060/binary/helloworld@mk3060.bin`，OTA升级用的是`out/helloworld@mk3060/binary/helloworld@mk3060.ota.bin`

Torchwoods, 2019-04-16:
> 
> 
> @Torchwoods rel2.1.0的helloworld@mk3060能正常运行的，你烧录的是哪个binary呢？正常的是`out/helloworld@mk3060/binary/helloworld@mk3060.bin`，OTA升级用的是`out/helloworld@mk3060/binary/helloworld@mk3060.ota.bin`

两个都烧过，通过串口烧录的
write 0x13200
主要是烧录后，在两个串口都没有日志输出。
你的工程修改了哪里？

xiaowenxia, 2019-04-16:
@Torchwoods 工程没有修改过，就按照默认的编译烧录就好了。

Torchwoods, 2019-04-16:
> 
> 
> @Torchwoods 工程没有修改过，就按照默认的编译烧录就好了。

aos make helloworld@mk3060 -c config && aos make

xiaowenxia, 2019-04-16:
> aos make helloworld@mk3060 -c config && aos make

是的。

Torchwoods, 2019-04-16:
> 
> 
> > aos make helloworld@mk3060 -c config && aos make
> 
> 是的。
烧录out/helloworld@mk3060/binary/helloworld@mk3060.bin日志正常，烧录OTA升级用的固件
out/helloworld@mk3060/binary/helloworld@mk3060.ota.bin，日志没有输出。

xiaowenxia, 2019-04-16:
@Torchwoods 是这样的，`helloworld@mk3060.ota.bin` 只是用于ota升级，直接烧录到0x13200无法正常运行的。

Torchwoods, 2019-04-16:
> 
> 
> @Torchwoods 是这样的，`helloworld@mk3060.ota.bin` 只是用于ota升级，直接烧录到0x13200无法正常运行的。

这个版本改了吗？以前好像都是烧录的xxx.ota.bin文件。手册里面也是这样描述的。

Torchwoods, 2019-04-16:
> 
> 
> @Torchwoods 是这样的，`helloworld@mk3060.ota.bin` 只是用于ota升级，直接烧录到0x13200无法正常运行的。

也就是说，xxx.ota.bin文件在设备启动OTA升级的时候，从服务器下载到设备，然后更新本地的固件。

xiaowenxia, 2019-04-16:
是的，`out/helloworld@mk3060/binary/readme.txt`里面有说明。

closed 2019-04-16 by Torchwoods