alibaba/AliOS-Things #951: Ubuntu下使用vscode调试example下的helloworld程序，一直报错，请求答疑，谢谢～

Steve-Kung, 2019-04-15:
**Describe the bug**
初次使用Ubuntu下使用vscode调试example下的helloworld程序，程序按照教程能够编译，但是调试选择之后报错。想请问一下问题出在哪。感谢

**environment**
1. developer(version): 
2. Ubuntu
3. example: 

**Expected behavior**

**Screenshots**
[image]
[image]

**Additional context**
我的launch,json文件如下
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug @ Mac/Linux",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceRoot}/out/helloworld@developerkit/binary/helloworld@developerkit.elf",
            "args": [],
            "stopAtEntry": true,
            "cwd": "${workspaceRoot}",
            "environment": [],
            "externalConsole": true,
            "miDebuggerServerAddress": "localhost:4242",
            "serverLaunchTimeout": 2000,
            "targetArchitecture": "ARM",
            "setupCommands": [
                {
                    "text": "cd ${workspaceRoot}"
                },
                {
                    "text": "source .gdbinit"
                },
                {
                    "text": "target remote localhost:4242"
                },
                {
                    "text": "file out/helloworld@developerkit/binary/helloworld@developerkit.elf"
                },
                {
                    "text": "break application_start"
                }
            ],
            "customLaunchSetupCommands": [],
            "launchCompleteCommand": "exec-run",
            "osx": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb"
            },
            "linux": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb"
            },
            "windows": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb.exe"
            }
        },
        {
            "name": "Debug @ Windows",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceRoot}/out/helloworld@developerkit/binary/helloworld@developerkit.elf",
            "args": [],
            "stopAtEntry": true,
            "cwd": "${workspaceRoot}",
            "environment": [],
            "externalConsole": true,
            "miDebuggerServerAddress": "localhost:4242",
            "serverLaunchTimeout": 2000,
            "targetArchitecture": "ARM",
            "setupCommands": [
                {
                    "text": "cd ${workspaceRoot}"
                },
                {
                    "text": "source .gdbinit"
                },
                {
                    "text": "target remote localhost:4242"
                },
                {
                    "text": "file out/helloworld@developerkit/binary/helloworld@developerkit.elf"
                },
                {
                    "text": "break application_start"
                }
            ],
            "customLaunchSetupCommands": [],
            "launchCompleteCommand": "exec-run",
            "osx": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb"
            },
            "linux": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb"
            },
            "windows": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb.exe"
            }
        },
        {
            "name": "CSKY Debug @ Windows/Linux",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceRoot}\\out\\helloworld@cb2201\\binary\\helloworld@cb2201.elf",
            "args": [],
            "stopAtEntry": true,
            "cwd": "${workspaceRoot}",
            "environment": [],
            "externalConsole": true,
            "miDebuggerServerAddress": "localhost:8888",
            "serverLaunchTimeout": 2000,
            "targetArchitecture": "X86_64",
            "setupCommands": [
                {
                    "text": "cd ${workspaceRoot}"
                },
                {
                    "text": "source .gdbinit"
                },
                {
                    "text": "target jtag jtag://localhost:8888"
                },
                {
                    "text": "file out/helloworld@cb2201/binary/helloworld@cb2201.elf"
                },
                {
                    "text": "set debug-in-rom"
                },
                {
                    "text": "hbreak application_start"
                }
            ],
            "customLaunchSetupCommands": [],
            "launchCompleteCommand": "exec-run",
            "linux": {
                "MIMode": "gdb",
                "miDebuggerPath": "csky-abiv2-elf-gdb"
            },
            "windows": {
                "MIMode": "gdb",
                "miDebuggerPath": "${workspaceRoot}\\build\\compiler\\gcc-csky-abiv2\\Win32\\bin\\csky-abiv2-elf-gdb.exe"
            }
        }
    ]
}

walker2048, 2019-04-15:
看样子是你没有在PATH设置好gdb的路径，直接调用肯定会有问题。build\compiler\gcc-arm-none-eabi\Linux64\bin

walker2048, 2019-04-15:
最佳的解决方案是直接把编译器路径添加到PATH中，平时命令行调用也比较舒服

TeemoRZ, 2019-04-16:
> 最佳的解决方案是直接把编译器路径添加到PATH中，平时命令行调用也比较舒服

[image]
我在windows下有同样情况，编译器路径设置如下：
"windows": {
                "miDebuggerPath": "${workspaceRoot}/build/compiler/gcc-arm-none-eabi/Win32/bin/arm-none-eabi-gdb"
            }

walker2048, 2019-04-16:
@TeemoRZ 试了一下。。。
确实不行，要不直接命令行GDB算啦？

xiaowenxia, 2019-04-16:
> > 最佳的解决方案是直接把编译器路径添加到PATH中，平时命令行调用也比较舒服
> 
> [image]
> 我在windows下有同样情况，编译器路径设置如下：
> "windows": {
> "miDebuggerPath": "${workspaceRoot}/build/compiler/gcc-arm-none-eabi/Win32/bin/arm-none-eabi-gdb"
> }

windows下的path需要用反斜杠，这样写：
```json
"windows": {
"miDebuggerPath": "${workspaceRoot}\\build\\compiler\\gcc-arm-none-eabi\\Win32\\bin\\arm-none-eabi-gdb.exe"
}
```

TeemoRZ, 2019-04-20:
> windows下的path需要用反斜杠，这样写：
>
> ```json
> "windows": {
> "miDebuggerPath": "${workspaceRoot}\\build\\compiler\\gcc-arm-none-eabi\\Win32\\bin\\arm-none-eabi-gdb.exe"
> }
> ```

正确答案

closed 2022-03-01 by skylarCai