alibaba/AliOS-Things #953: windows 10下developerkit helloworld 烧写出错

chq007, 2019-04-17:
按照example目录下面的helloworld里面的readme.md一步步进行操作，在烧写之前都没有报错。然后执行“aos upload”命令时出错，log信息如下：

[INFO]: Currently in aos_sdk_path: 'D:\Code\IoT\AliOS-Things'
[INFO]: Not set target, read target from .config
[INFO]: Target: helloworld@developerkit

scons: Reading SConscript files ...
[INFO]: Scons to upload!args:{'APPLICATION': 'helloworld', 'COMMAND': 'upload', 'BOARD': 'developerkit'}

[INFO]: Target: helloworld@developerkit
[INFO]: Currently in aos_sdk_path: 'D:\Code\IoT\AliOS-Things'

[INFO]: Running cmd:
        'D:\Code\IoT\AliOS-Things/build/cmd/win32/st-flash.exe --reset write D:\Code\IoT\AliOS-Things/out/helloworld@developerkit/binary/helloworld@developerkit.bin 0x08000000'st-flash 1.5.0
2019-04-17T23:14:44 WARN usb.c: Couldn't find any ST-Link/V2 devices
---host_os:Win32
[ERROR]: Firmware upload failed!

[AliOS-Things] ERROR: "scons" returned error code -1.
[AliOS-Things] ERROR: Command "scons -f D:\Code\IoT\AliOS-Things\build/ucube.py COMMAND=upload APPLICATION=helloworld BOARD=developerkit" in "D:\Code\IoT\AliOS-Things"

chq007, 2019-04-17:
从报错信息看貌似是无法找到st-link/v2 devices, 因此使用stm32 st-link工具测试是否能够连接，发现可以成功连接并看到设备内部内存信息，连接过程的log信息如下：

23:20:29 : ST-LINK SN : 066BFF554952728381134352
23:20:29 : ST-LINK Firmware version : V2J28M18
23:20:29 : Connected via SWD.
23:20:29 : SWD Frequency = 4,0 MHz.
23:20:29 : Connection mode : Normal.
23:20:29 : Debug in Low Power mode enabled.
23:20:29 : Device ID:0x461 
23:20:29 : Device flash Size : 1MBytes
23:20:29 : Device family :STM32L49x

个人感觉是vscode里面的st-flash.exe与stm32 st-link utility中使用的设备驱动程序不是同一个。本人菜鸟一个，不知道这种情况如何解决，希望大牛能够指点一下，不胜感谢。

walker2048, 2019-04-22:
建议截图一下设备管理界面，看看驱动是否正确安装，之前也有人遇到过驱动安装不正常的

xiaowenxia, 2019-04-22:
> 个人感觉是vscode里面的st-flash.exe与stm32 st-link utility中使用的设备驱动程序不是同一个。本人菜鸟一个，不知道这种情况如何解决，希望大牛能够指点一下，不胜感谢。

是用的同一个驱动程序。
是不是其他程序占用了st-link导致st-flash.exe无法识别呢？

zeng-hello, 2019-07-23:
请问您现在解决了吗

closed 2022-03-01 by skylarCai