alibaba/AliOS-Things #955: vs code编译 helloworld错误

wollir, 2019-04-24:
请问这个是什么问题啊，代码没改，环境d搭建问题？？
> Executing task: aos make <
aos-cube version: 0.3.3
Check if required tools for developerkit exist
ECHO 处于关闭状态。                                                                                                                        
Build C
onfiguration:                                                                                                                       ====================                                                                                                                       App:
Build Configuration:
====================App:   helloworld
Board: developerkit
====================
ECHO 处于关闭状态。                                                                                                                        
Build AOS Now                                                                                                                              TOOLCHAIN_PATH=                                                                                                                            make.exe[1]: *** No rule to make target 'osal/./include/aos/aos.h', needed by 'out/helloworld@developerkit/Modules/app/example/helloworld/helloworld.o'.  Stop.                                                                                                                       make.exe: *** [build/Makefile:395: main_app] Error 2                                                                                       [AliOS-Things] ERROR: "I:/alios/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.                                              [AliOS-Things] ERROR: Command "I:/alios/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=I:/alios/AliOS-Things/build -e -f build/Makefile" in "I:\alios\AliOS-Things"

walker2048, 2019-04-24:
建议先试试aos make clean清理编译垃圾

wollir, 2019-04-24:
> 建议先试试aos make clean清理编译垃圾

刚刚用这个方法解决了，多谢

closed 2022-03-01 by skylarCai