alibaba/AliOS-Things #958: 用mk3060跑2.1.0的版本的helloworld出现问题

Ollieab1, 2019-04-26:
**Describe the bug**
用mk3060跑2.1.0的版本的helloworld串口没有打印相应的数据，下载完程序之后，我已经重启过开发板，但用2.0.0的版本跑这个历程就能跑成功，是2.1.0的需要在某个地方修改什么配置吗？

**environment**
1. branch(version): 2.1.0
2. Desktop OS(windows linux or macOS):windows
3. example: helloworld

**Expected behavior**
希望能够回复和解决下这个问题

**Screenshots**
这个是2.1.0版本的截图
[image]
这个是2.1.0版本编译好之后下载调试界面
[image]
这个是2.0.0版本的截图
[image]
这个是2.0.0版本编译好之后下载调试界面
[image]

**Additional context**

xiaowenxia, 2019-04-26:
2.1的代码烧录的是`helloworld@mk3060.bin`，不是`helloworld@mk3060.ota.bin`哈。

Ollieab1, 2019-04-26:
> 2.1的代码烧录的是`helloworld@mk3060.bin`，不是`helloworld@mk3060.ota.bin`哈。

卧槽，搞了一个晚上+一个早上，终于解决了

xiaowenxia, 2019-04-26:
@Ollieab1 helloworld@mk3060.bin 同目录下面有readme的，还是要看一下的。

Ollieab1, 2019-04-26:
> @Ollieab1 [helloworld@mk3060.bin](mailto:helloworld@mk3060.bin) 同目录下面有readme的，还是要看一下的。

readme看了，可能没看仔细

closed 2022-03-01 by skylarCai