alibaba/AliOS-Things #960: wiki some error

letfly, 2019-04-29:
1. https://github.com/alibaba/AliOS-Things/wiki/MK3060-Hardware-Setup
./build/OpenOCD/Linux64/openocd --> ./build/OpenOCD/Linux64/bin/openocd
2. https://github.com/alibaba/AliOS-Things/wiki/Quick-Start
aos-cube 0.3.x required之前需要aos make menuconfig
3. pip 用easy_install 替代

xiaowenxia, 2019-04-29:
@letfly 谢谢你的纠正:+1: 。

> aos-cube 0.3.x required之前需要aos make menuconfig

这个需要说明一下哈，AliOS Things 2.1代码可以使用 `aos make menuconfig`来进行手动配置，也可以使用`aos make  app@board -c config` 进行一键配置（使用默认配置），文档里面使用的是一键配置。

> pip 用easy_install 替代

为啥推荐用easy_install 替代pip呢？

letfly, 2019-04-30:
@xiaowenxia 

> 这个需要说明一下哈，AliOS Things 2.1代码可以使用 aos make menuconfig来进行手动配置，也可以使用aos make app@board -c config 进行一键配置（使用默认配置），文档里面使用的是一键配置。
刚测试了下，-c config可以

> pip 用easy_install 替代
ubuntu16下默认配置easy_install的一些源更多

closed 2019-04-30 by letfly

xiaowenxia, 2019-04-30:
@letfly 这个倒无所谓，只要能够下载aos-cube就行哈。