alibaba/AliOS-Things #961: mk3060配网问题

Ollieab1, 2019-04-29:
**Is your feature request related to a problem? Please describe.**
用mk3060进行配网，死在了图中的那一步，大佬们这是哪里出问题了？

**Describe the solution you'd like**
[image]

**Additional context**

xiaowenxia, 2019-04-29:
@Ollieab1  有详细log么？把详细log发出来估计更好找问题。

Ollieab1, 2019-04-29:
> @Ollieab1 有详细log么？把详细log发出来估计更好找问题。

t0 init success

Press key 'w' into 2nd boot cmd, Waiting 100ms ...
Bootup flag 0x0, num 0x0
flash txpwr table:0x7
dif g and n20 ID in flash:4
read txpwr tab from flash success
DPLL Unlock
enter low level!
mac fc:ee:e6: 5:72:b5
leave low level!
app_init finished
start-----------hal
             Welcome to AliOS Things           
[prt] log level set as: [ 5 ]
start_netmgr
IOTX_AWSS_START[002006]<I> zconfig_init

Soft_AP_start
[002018]<F> chan 1

[002066]<F> [1] ssid:B-LINK_885B58, mac:20f41b885b58, chn:3, rssi:-1, adha:0

[002114]<F> [2] ssid:Apple Watch, mac:6cb0ce3b5a8f, chn:2, rssi:-1, adha:0

[002148]<F> [3] ssid:CMCC-WEB, mac:3ce5a69ab6d3, chn:1, rssi:-1, adha:0

[002150]<F> [4] ssid:CMCC-EDU, mac:3ce5a69ab6d4, chn:1, rssi:-1, adha:0

[002272]<F> chan 6

[002294]<F> [5] ssid:ZUCC-CMCC, mac:3ce5a69aa6f2, chn:6, rssi:-1, adha:0

[002294]<F> [6] ssid:CMCC-WEB, mac:3ce5a69aa6f3, chn:6, rssi:-1, adha:0

[002334]<F> [7] ssid:chestnut, mac:747d24910020, chn:6, rssi:-1, adha:0

[002350]<F> [8] ssid:刮开获得密码▇▇▇, mac:2cb21a4491a2, chn:6, rssi:-1, adha:0

[002498]<F> [9] ssid:CMCC-EDU, mac:3ce5a69aa6f4, chn:6, rssi:-1, adha:0

[002526]<F> chan 11

[002540]<F> [10] ssid:101, mac:2c56dcf473bc, chn:12, rssi:-1, adha:0

[002556]<F> [11] ssid:爸爸, mac:3c3300d5255a, chn:12, rssi:-1, adha:0

[002560]<F> [12] ssid: 免费安全共享WiFi, mac:3e3300d5255a, chn:12, rssi:-1, adha:0

[002574]<F> [13] ssid:CMCC-NET, mac:8cab8e8a7270, chn:12, rssi:-1, adha:0

[002610]<F> [14] ssid:LOT, mac:b66d839dec08, chn:11, rssi:-1, adha:0

[002616]<F> [15] ssid:old hu, mac:d2577bf1f360, chn:11, rssi:-1, adha:0

[002780]<F> chan 1

[002808]<F> [16] ssid:ZUCC-CTC, mac:3822d6a8bf31, chn:1, rssi:-1, adha:0

[002968]<F> [17] ssid:ZUCC-CMCC, mac:3ce5a69ab6d2, chn:1, rssi:-1, adha:0

[003034]<F> chan 2

[003068]<V> awss config press 2

进入配网
do_awss_active 0
[003068]<F> enable awss
[... 106 lines cut ...]
add extral movement in test
[011762]<F> sending succeeded.
[011762]<F> sending succeeded.
[011762]<F> sending succeeded.
[011762]<F> sending succeeded.
[011764]<F> sending succeeded.
[012764]<F> connect 'LOT'
[012764]<D> bssid: ffffffb6:6d:ffffff83:ffffff9d:ffffffec:08
[012764]<F> switch to ap: 'LOT'
------------------------suspend station
sta_ip_down
Cancelling scan request
SM_DISCONNECT_IND
wpa_dInit
wpa_supplicant_req_scan
Setting scan request: 1.100000 sec
MANUAL_SCAN_REQ
wpa_supplicant_scan
wpa_drv_scan
wpa_send_scan_req
wpa_driver_scan_cb
wpa_get_scan_rst:1
wpa_supplicant_connect
Cancelling scan request
wpa_driver_associate

closed 2019-04-30 by Ollieab1