alibaba/AliOS-Things #965: win10下vs code 编译helloworld不过

jiangzhi18, 2019-05-05:
App:   helloworld
Board: sv6266_evb

ECHO 处于关闭状态。
Creating out/config/autoconf.h ...
Making config file for first time
app_version:app-1.0.0-20190505.1705
build/build_rules/toolchain/aos_toolchain_nds32le-elf-newlib-v3.mk:32: *** incorrect 'make' used ("E:/AliOS-Things/build/cmd/win32/make.exe") - please use:  (Windows) .\make.exe <target_string>    (OS X, Linux) ./make <target_string>.  Stop.
make.exe: *** [build/Makefile:344: out/helloworld@sv6266_evb/config.mk] Error 2
[AliOS-Things] ERROR: "E:/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/AliOS-Things/build -e -f build/Makefile" in "E:\AliOS-Things"

xiaowenxia, 2019-05-06:
sv6266_evb 好像只能在linux下面编译，试试在linux下面编译哈。

WKDGYB, 2019-05-20:
我也这样 楼主你解决了吗

purecafe2016, 2019-08-22:
然后屏就没有反应了，还有救吗？

WKDGYB, 2019-08-23:
还是考虑到 环境没装好  aos-cube 我之前吧这个完全删掉再装 就好了

WKDGYB, 2019-08-23:
如果你说是板子上的屏幕的话 有一个初始化的bin文件 可以恢复

786822394@qq.com
 
发件人： purecafe2016
发送时间： 2019-08-23 05:42
收件人： alibaba/AliOS-Things
抄送： WKDGYB; Comment
主题： Re: [alibaba/AliOS-Things] win10下vs code 编译helloworld不过 (#965)
然后屏就没有反应了，还有救吗？

Lign17314, 2020-04-21:
只能在linux下编译，要提前安装好下面这些
$ sudo apt-get install zlib1g:i386 
$ sudo apt-get install lib32z1  
$ sudo apt-get install lib32ncurses5 
$ sudo apt-get install e2fslibs:i386 
$ sudo apt-get install gcc-multilib
$ sudo apt-get install g++-multilib