alibaba/AliOS-Things #974: [SoC2019]Application development framework

Hunter1990, 2019-05-09:
- 描述
  普通的嵌入式开发多通过第三方IDE工具，使用芯片厂商发布的代码包或代码生成工具，生成相应启动代码及bsp，用户需自行移植需要的第三方软件，如果是IoT系统，还需移植网络连接sdk以及和云端能力相关的设备端实现，一起进行集成开发和调试，越长的开发链路会消耗越多的人力成本和时间。因此我们需要更加高效的开发方式。
- Description
Ordinarily, most software development on embedded systems relies on third-party IDE tools, using code package or code generation tool released by chip/module vendors to generate corresponding startup code and BSP, with which users need to transplant the required third-party softwares. For IoT applications, it further needs to transplant the network-connecting sdk and device-side implementations of cloud capabilities for integrated development and debugging. The longer the development link, the more labor and time it consumes. So we need a more efficient development approach.

- 目标
  通过(不限于)以下一种或多种途径，让基于 AliOS Things 的应用开发更简单、更高效，更适合云端一体化的 IoT 项目。
- Goal
To make application development based on AliOS Things easier and more efficient, and more suitable for cloud-integrated IoT projects, by (but not limited to) one or more of the following approaches.

- 参考
  方向1: 兼容业界优秀的嵌入式应用开发框架，API兼容即可。如：使 Arduino 应用和库代码可以在 AliOS Things 快速移植并运行。
  方向2: 支持作为第三方framework集成到[PlatformIO](https://platformio.org/frameworks)，使得用户可以通过PlatformIO工具基于AliOS Things开发。
  方向3：利用脚本引擎加速应用开发，完善脚本引擎底层硬件资源映射及相关工具开发。如 [MicroPython](https://github.com/alibaba/AliOS-Things/tree/rel_2.1.0/3rdparty/experimental/micropython) 或 [TinyEngine](https://github.com/alibaba/AliOS-Things/tree/rel_2.0.0/middleware/tinyengine)
- Reference
[1] Make AliOS Things compatible with the industry's excellent embedded application development framework, API compatibility is enough. For example, make porting Arduino application and library code expeditious on AliOS Things.
[2] Support for integration as a third-party framework on PlatformIO, enabling users to develop on AliOS Things via the PlatformIO tool.
[3] Use script engines to accelerate application development, improving mapping of hardware resources and development of related tools. For example,
[MicroPython](https://github.com/alibaba/AliOS-Things/tree/rel_2.1.0/3rdparty/experimental/micropython) and [TinyEngine](https://github.com/alibaba/AliOS-Things/tree/rel_2.0.0/middleware/tinyengine).

starcruizer, 2019-05-12:
I would like to contribute to this Idea for this summers

Hunter1990, 2019-05-17:
> I would like to contribute to this Idea for this summers

Could you please describe your idea or proposal briefly?

Hunter1990, 2019-05-17:
> I would like to contribute to this Idea for this summers

@starcruizer Could you please describe your idea or proposal briefly?

MRNIU, 2019-05-28:
请问方向一中，[arduino 库](https://www.arduino.cc/en/Reference/Libraries)的移植是仅需要移植标准库，还是全部都要移植？

librae8226, 2019-05-28:
@MRNIU 没有限制，可以选择性移植(优先考虑用得多的 library)。这个方向主要是对 arduino 框架做兼容和支持。

starcruizer, 2019-05-28:
> > I would like to contribute to this Idea for this summers
> 
> @starcruizer Could you please describe your idea or proposal briefly?

Sure, can you please provide me your email?

Hunter1990, 2019-05-28:
> > > I would like to contribute to this Idea for this summers
> > 
> > 
> > @starcruizer Could you please describe your idea or proposal briefly?
> 
> Sure, can you please provide me your email?

Please submit on this link https://page.aliyun.com/form/act1008956472/index.htm?spm=5176.12783092.1351754.1.247755faZYc68b

MRNIU, 2019-05-29:
我刚刚完成移植 Arduino 库的提案，但是提交的时候才发现我已经把两次提交的机会用完了。。。我在官方网站没有找到撤销的相关选项。。请问有什么其它的方法吗？

librae8226, 2019-05-30:
@MRNIU 再提交一次你的最新版本哈

MRNIU, 2019-05-30:
@librae8226 
你好，我已在网站再次尝试提交，依然提示我提交次数受限
[image]

closed 2022-03-01 by skylarCai