alibaba/AliOS-Things #975: [SoC2019]Customized firmware based on AliOS Things for connecting Alibaba Cloud

Hunter1990, 2019-05-09:
- 描述
  业界常见的连接模组(WiFi或广域网)通过各类协议如AT命令(UART)或定制协议(SPI/SDIO)接入互联网，而具体要接入阿里云，仍需要在主控MCU侧进行sdk移植和云端进行鉴权和后续交互，于是此处增加了开发复杂度，也占用了主控MCU更多资源。所以如果有一套为模组量身定制的标准固件，用户拿到以后可以专注于业务逻辑开发，而不用担心连接和交互，会很大程度提高开发效率，缩短产品上市时间。
- Description
Common network-connecting modules (WiFi or WAN) in industry access Internet through various protocols such as AT commands (UART) or custom protocols (SPI/SDIO). To access Alibaba Cloud, it still needs to perform sdk transplanting on the master MCU side, thus increasing development complexity and occupying more resources of the master MCU. So if you have a standard firmware tailored for the module, users can focus on business logic development without worrying about connectivity and interaction, which will greatly improve development efficiency and shorten time to market.

- 目标
  为模组定制基于AliOS Things的固件，在上电以后简单配置即可接入阿里云进行数据交互，从而更快在设备侧集成阿里云各类产品能力。
- Goal
To customize firmwares based on AliOS Things for modules. After power-on, you can access Alibaba Cloud for data interaction with simple configuration, thus integrating Alibaba Cloud's various capabilities on device side.

- 参考
  固件采用AliOS Things作为开发基座，通过为用户提供AT指令的支持实现用户业务逻辑接口。同时，通过模组提供的GPIO引脚，为用户提供硬件的模组状态指示，如连接状态，故障状态，以及工作模式的切换等。
  AT固件指令集设计：启动自动配网，静态配置AP信息，上电自动连接AP开关，设置串口参数，DHCP使能，静态配置ip信息，配置连接信息，连接服务器，向云端发送数据等。
- Reference
The firmware uses AliOS Things as the development base to provide users with AT commands as interfaces for implementing business logics. At the same time, through the GPIO pin provided by the module, users are provided with hardware module status indications, such as connection status, fault status, and switching of working mode.
AT firmware instructions set design: start automaticly distributed network, statically configure AP information, automatically connect to AP switch, set serial port parameters, enable DHCP, statically configure IP information, configure connection information, connect server, send data to cloud, etc.

ysppp, 2019-05-23:
这个issues的意思是为特定的硬件模组如esp8266这样的硬件设备编写基于AT指令的API提供给用户调用吗？

Hunter1990, 2019-05-24:
> 这个issues的意思是为特定的硬件模组如esp8266这样的硬件设备编写基于AT指令的API提供给用户调用吗？

这个issue的核心思想就是提供一个串口和一套AT，MCU继续只专注于业务和应用，连云方面只需通过AT由模组完成。

ysppp, 2019-05-24:
> > 这个issues的意思是为特定的硬件模组如esp8266这样的硬件设备编写基于AT指令的API提供给用户调用吗？
> 
> 这个issue的核心思想就是提供一个串口和一套AT，MCU继续只专注于业务和应用，连云方面只需通过AT由模组完成。

我阅读了https://help.aliyun.com/document_detail/111856.html?spm=a2c4g.11186623.6.569.45ea1fb6DiFgxx 这篇帮助文档，联系到这个issue，是否意味着完成模组商要做的AT指令相关的工作即可？

Hunter1990, 2019-05-24:
> > > 这个issues的意思是为特定的硬件模组如esp8266这样的硬件设备编写基于AT指令的API提供给用户调用吗？
> > 
> > 
> > 这个issue的核心思想就是提供一个串口和一套AT，MCU继续只专注于业务和应用，连云方面只需通过AT由模组完成。
> 
> 我阅读了https://help.aliyun.com/document_detail/111856.html?spm=a2c4g.11186623.6.569.45ea1fb6DiFgxx 这篇帮助文档，联系到这个issue，是否意味着完成模组商要做的AT指令相关的工作即可？

表面看来是只要做一套

Hunter1990, 2019-05-24:
> > > 这个issues的意思是为特定的硬件模组如esp8266这样的硬件设备编写基于AT指令的API提供给用户调用吗？
> > 
> > 
> > 这个issue的核心思想就是提供一个串口和一套AT，MCU继续只专注于业务和应用，连云方面只需通过AT由模组完成。
> 
> 我阅读了https://help.aliyun.com/document_detail/111856.html?spm=a2c4g.11186623.6.569.45ea1fb6DiFgxx 这篇帮助文档，联系到这个issue，是否意味着完成模组商要做的AT指令相关的工作即可？

如何简化连云，甚至做到无缝连云，是这个issue的核心

stephenmasih, 2019-05-30:
Hello sir, I want to contribute to this project in Alibaba SOC 2019. I want to build application firmware in esp8266, to connect and interact with Alibaba cloud.

Hunter1990, 2019-06-05:
> Hello sir, I want to contribute to this project in Alibaba SOC 2019. I want to build application firmware in esp8266, to connect and interact with Alibaba cloud.

@stephenmasih Appreciate and welcome! Please commit your proposal on this link https://developer.aliyun.com/special/summerofcode2019en

ronnie14165, 2019-06-20:
项目负责人您好：
       我曾提交了该项目的提案并于昨天下午收到了贵公司的电话，简单交流后，负责人表示想要看看我之前的项目，说会在稍后发一个邮箱地址到我的邮箱，让我将相关内容发送过去。昨晚我准备好了相关内容，但至今并未受到贵公司的邮件。是否可以请您再发一次呢，我的邮箱是1416558969@qq.com或ml62@hw.ac.uk
      我非常珍惜这次机会，也希望我有能力胜任
      谢谢！

closed 2022-03-01 by skylarCai