alibaba/AliOS-Things #977: [SoC2019]GUI development tool for AliOS Things

Hunter1990, 2019-05-09:
- 描述
  AliOS Things基于littlevGL集成了完整的GUI的开发框架，但目前的GUI开发只能通过编写代码方式，如添加控件、增加文字、设置背景图片、页面布局等都是通过直接调用相应的API实现。另外，目前已实现通过调用sdl2库在PC上模拟开发。
  GUI代码：3rdparty/experimental/gui/littlevGL
  样例：app/example/dk/dk_gui
  开发指南：https://yq.aliyun.com/articles/582795
- Description
AliOS Things integrates a complete GUI development framework based on littlevGL, but current GUI development can only be achieved by writing code, such as adding controls, adding text, setting background images, page layout, etc. by directly calling the corresponding APIs. In addition, a simulating tool on PC has been implemented by calling the sdl2 library.
GUI code: 3rdparty/experimental/gui/littlevGL
Example: app/example/dk/dk_gui
Guide: https://yq.aliyun.com/articles/582795

- 目标
  开发一套可视化开发工具，本地或者web端均可，实现GUI的可视化开发，大致包含如下功能：
  1.拖拽式的添加控件，设置控件位置、大小和各种属性（已有能力移植和集成）
  2.界面上添加文字以及设置字体（需要附带生成相应字库）
  3.实现各个组件进入、切换、退出效果的设置
  4.可以在界面上编写事件处理代码（触摸、拖动）
  5.在PC上运行并查看效果
  6.直接生成C代码，结合嵌入式设备驱动即可运行
- Goal
To develop a set of visual development tools, either locally or on the web, to realize visual development of GUI, which roughly includes the following functions:
  1. Drag and drop to add controls, to set the position, size and various properties of the control (port and integrate)
  2. Add text and set font on the interface (requires the corresponding font to be generated)
  3. Realize settings of each component to enter, switch, and exit effects
  4. Possibility to write event processing code (touch, drag) on the interface
  5. Run on the PC to verify the effects
  6. Directly generate C code, combined with embedded device drivers to run

gootoomoon, 2019-05-17:
期待,  希望能实现 TouchGFX  一样丰富的控件, 或者控件部分可以开源出来让所有人来添加修改一起完善这个工程, 目前MCU+小屏的应用越来越多了, 非常需要这种 GUI Designer 工具的支持

Hunter1990, 2019-05-23:
> 期待, 希望能实现 TouchGFX 一样丰富的控件, 或者控件部分可以开源出来让所有人来添加修改一起完善这个工程, 目前MCU+小屏的应用越来越多了, 非常需要这种 GUI Designer 工具的支持

如果有兴趣，非常欢迎提出你的proposal或者idea

xiaoerlaigeid, 2019-05-23:
你好，我对这个项目比较感兴趣想问下这个环境是可以运行在虚拟机里吗？我安装AliOS Things时候出现了
`E: 无法定位软件包 libssl-dev:i386
E: 无法定位软件包 libncurses-dev:i386
E: 无法定位软件包 libreadline-dev:i386`
是必须要用基于arm的嵌入式板才可以吗？一般开发有什么推荐的系统套件吗？

xiaowenxia, 2019-05-23:
@xiaoerlaigeid 
> E: 无法定位软件包 libssl-dev:i386 E: 无法定位软件包 libncurses-dev:i386 E: 无法定位软件包 libreadline-dev:i386

32位的linux可能出现这个问题。

> 必须要用基于arm的嵌入式板才可以吗？一般开发有什么推荐的系统套件吗？

不是必须的。`AliOS Things`支持的board可以参考`board/`目录。

gootoomoon, 2019-06-18:
项目什么时候可以开始? 有具体的项目开发或者发布地址吗? 想尽快参与测试熟悉起来

Hunter1990, 2019-06-18:
> 项目什么时候可以开始? 有具体的项目开发或者发布地址吗? 想尽快参与测试熟悉起来

你好，报名已结束，目前正在评审proposal

kisvegabor, 2019-07-23:
Hi,

I'm the author of LittlevGL and happy to see that you are interested in a GUI designer. 

We've already started to work on GUI designer but it's paused right now in lack of time.
Here is what we have so far: https://github.com/littlevgl/lv_gui_builder/issues/1#issuecomment-495178918

jason-sophia, 2019-07-30:
@kisvegabor 
Hi Gabor,
We believe that a good graphical development environment is necessary for GUI development. We have already started this part of the work a month ago, and now we have completed part of it. We don't know that you are also developing this feature. I think we can develop independently, then we can choose a better one as an official recommendation. Is that OK for you ?

kisvegabor, 2019-07-30:
> I think we can develop independently, then we can choose a better one as an official recommendation. Is that OK for you ?

Ok. 
Can you show some preliminary screenshots?

cc @embeddedt

embeddedt, 2019-07-30:
Also, what license are you planning to release this GUI designer under?

closed 2022-03-01 by skylarCai