alibaba/AliOS-Things #978: 2019 Alibaba Summer of Code

Hunter1990, 2019-05-09:
Welcome to the open source world!If you haven't planned how to spend this summer, come to the Alibaba Summer of Code and coding with us! 💻

Alibaba Summer of Code is a global program focused on engaging students directly into open source software development. Under the guidance of the mentor in the Alibaba open source project, students experience the software development in the real world.Alibaba Summer of code will begin between June 28th and September 20th. Students can use the summer time to participate in the open source project and work with the core members of the project.

# What you can get?
On this exclusive developer journey, students will have the opportunity to:

- Participate in the top projects of the International Open Source Foundation;
- Get a scholarship from Alibaba;
- Obtain an open source contributor certificate;
- Get a fast pass of Alibaba Internship/recruitment;
- Get your code adopted and used by the open source project！

# Our Mentor
Yiran Liao
GitHub ID: librae8226
Alibaba Cloud IoT expert

# Apply Now! 
You can click on this link to apply directly: https://github.com/alibaba/AliOS-Things/issues?q=is%3Aissue+is%3Aopen+label%3ASoC2019
full list of Alibaba Summer of Code Projects：https://developer.aliyun.com/summerofcode2019

# Timeline⏳
[image]

# Contact us!
Email Address: aliopensource@alibabacloud.com

MRNIU, 2019-05-24:
请问我可以提出自己的 idea 吗？
我想为 AliOS-Things 移植新的文件系统比如 tmpfs。

Hunter1990, 2019-05-24:
> 请问我可以提出自己的 idea 吗？
> 我想为 AliOS-Things 移植新的文件系统比如 tmpfs。

提出自己的idea同样也是欢迎的，但在提出proposal时需要详细阐述

MRNIU, 2019-05-27:
@Hunter1990 
我已提交一篇关于为 AliOS-Things 实现 ext2 文件系统的 proposal，请问是否与贵项目开发人员有撞车风险？

QianTony, 2019-05-27:
> @Hunter1990
> 我已提交一篇关于为 AliOS-Things 实现 ext2 文件系统的 proposal，请问是否与贵项目开发人员有撞车风险？

EXT系列的文件系统是基于块设备的， 如果移植到RTOS上需要考虑在裸flash上的情况（需要考虑磨损平衡），而且ext2是索引式文件系统可靠性没有ext4那种日志型的高（两者占用系统资源都挺高的）

Hunter1990, 2019-05-27:
> @Hunter1990
> 我已提交一篇关于为 AliOS-Things 实现 ext2 文件系统的 proposal，请问是否与贵项目开发人员有撞车风险？

我们暂时没有开发ext2文件系统的计划，如上一条我同事所述，需思考得完善一些

MRNIU, 2019-05-28:
我本来是想做 fat 的，但是发现贵方已经写好了。文件系统方向似乎没有适合我的 idea 了。。。我会再考虑一下的，谢谢

wenjinglee1104, 2019-05-29:
你好，我想要为AliOS做出贡献。今年申请过GSoC,但是准备时间过于仓促。不过学习到了写PR需要和导师沟通。请问如果我想申请写PR的话，需要了解哪方面的知识呢？

Hunter1990, 2019-06-05:
> 你好，我想要为AliOS做出贡献。今年申请过GSoC,但是准备时间过于仓促。不过学习到了写PR需要和导师沟通。请问如果我想申请写PR的话，需要了解哪方面的知识呢？

@wenjinglee1104 这个项目的PR跟Github遵循标准流程，请参考https://help.github.com/en/articles/creating-a-pull-request

closed 2022-03-01 by skylarCai