alibaba/AliOS-Things #990: 使用手机热点配网一直失败

jsonsiyuan, 2019-05-28:
**Describe the bug**

使用手机热点配网一直失败
**environment**
1. version: 2.1.0
2. chip or mode：MK3080 and ESP8266

**Expected behavior**
just like smartconfig

**running log**
nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05
empty
Wifi init success!!
mode : sta(68:c6:3a:b2:6d:0b)
add if0
             Welcome to AliOS Things           
[prt] log level set as: [ 5 ]
[000010]<V> start_netmgr

[002020]<V> do_awss_active 0

[002020]<F> enable awss

[002020]<V> ##########IOTX_AWSS_ENABLE
[002020]<V> ###########IOTX_AWSS_START
[002020]<I> zconfig_init

[002040]<F> chan 1

[002050]<F> [1] ssid:TP-LINK_C111, mac:7844fd90c111, chn:1, rssi:-85, adha:0

[002080]<F> [2] ssid:Dooya_SmartMotor, mac:50fa849209c6, chn:1, rssi:-87, adha:0

[002090]<F> [3] ssid:Dlinkwyy, mac:c8d3a3185c1e, chn:1, rssi:-90, adha:0

[002090]<F> [4] ssid:PiaoHong-Wifi, mac:52fa84a209c6, chn:1, rssi:-86, adha:0

[002110]<F> [5] ssid:TP-LINK_5A7F, mac:f631c3b885fc, chn:1, rssi:-96, adha:0

[002130]<F> [6] ssid:ganyecheng, mac:c8d3a3185c22, chn:3, rssi:-67, adha:0

[002310]<F> chan 6

[002350]<F> [7] ssid:RootMesh, mac:1cf29acadce8, chn:6, rssi:-75, adha:0

[002370]<F> [8] ssid:TP-LINK_IKEA, mac:94d9b32f8429, chn:6, rssi:-74, adha:0

[002390]<F> [9] ssid:dd-wrt, mac:bcaec5e7b4dc, chn:6, rssi:-87, adha:0

[002410]<F> [10] ssid:cisco-3B4B, mac:d824bd773b4d, chn:6, rssi:-92, adha:0

[002420]<F> [11] ssid:demo, mac:487adab296b0, chn:6, rssi:-91, adha:0

[002490]<F> [12] ssid:360免费WiFi-B7, mac:008736234db7, chn:6, rssi:-95, adha:0

[002580]<F> chan 11

[002590]<F> [13] ssid:H3C, mac:b0f963481f90, chn:11, rssi:-88, adha:0

[002590]<F> [14] ssid:Dooya, mac:70f96db63bf0, chn:11, rssi:-84, adha:0

[002610]<F> [15] ssid:office, mac:70f96dd14bf0, chn:11, rssi:-76, adha:0

[002640]<F> [16] ssid:zhuqiqi, mac:224ce59a625d, chn:11, rssi:-81, adha:0

[002650]<F> [17] ssid:RootMesh, mac:1cf29acafb80, chn:11, rssi:-82, adha:0

[002660]<F> [18] ssid:kaohe1, mac:206be75a225a, chn:11, rssi:-65, adha:0

[002660]<F> [19] ssid:TP-LINK_3596, mac:7405a5263596, chn:11, rssi:-75, adha:0

[002680]<F> [20] ssid:RootMesh, mac:1cf29acabea8, chn:11, rssi:-85, adha:0

[002850]<F> chan 1

[002860]<F> [21] ssid:Moorgen_Hotel, mac:141fba8b14a8, chn:1, rssi:-93, adha:0

[003120]<F> chan 2

[003120]<F> [22] ssid:DYWM_TEST02, mac:50642b4685d5, chn:3, rssi:-93, adha:0

[003270]<F> [23] ssid:???-TEST, mac:ec4118429898, chn:2, rssi:-92, adha:0

[003390]<F> chan 3

[003660]<F> chan 4

[003670]<F> [24] ssid:Dooya_PH, mac:8cab8ed7c838, chn:4, rssi:-79, adha:0

[003930]<F> chan 5

[004200]<F> chan 6

[004470]<F> chan 7

[004480]<F> [25] ssid:D-Link_DIR-619L, mac:b0c554829c18, chn:8, rssi:-81, adha:0

[004740]<F> chan 8

[005010]<F> chan 9

[005280]<F> chan 10

[005550]<F> chan 11

[005820]<F> chan 12

[005820]<F> [26] ssid:TP-LINK_6C10, mac:ec26ca986c10, chn:12, rssi:-95, adha:0

[006090]<F> chan 13

[006360]<F> chan 1

[006630]<F> chan 2

[006900]<F> chan 3

[007170]<F> chan 4

[007440]<F> chan 6

[007500]<F> [27] ssid:ID#06, mac:b8621f537d1d, chn:6, rssi:-96, adha:0

[007710]<F> chan 8

[007980]<F> chan 11

[008250]<F> chan 12

[008520]<F> chan 1

[008790]<F> chan 6

[009060]<F> chan 11

[009330]<F> chan 1

[009600]<F> chan 2

[009870]<F> chan 3

[010140]<F> chan 4

[010410]<F> chan 5

[010680]<F> chan 6

[010950]<F> chan 7

[011220]<F> chan 8

[011490]<F> chan 9

[011760]<F> chan 10

[012030]<F> chan 11

[012300]<F> chan 12

[012570]<F> chan 13

[012840]<F> chan 1

[013110]<F> chan 2

[013120]<F> [28] ssid:aha, mac:482ca0052659, chn:3, rssi:-50, adha:1

############sun [awss_recv_callback_aha_ssid]
[013130]<D> found default ssid: aha

[013130]<F> chan lock @ 2

[013130]<V> #############IOTX_AWSS_LOCK_CHAN
[013130]<F> zconfig done. ssid:aha, mac:482ca0052659

[013130]<D> ssid:aha, bssid:482ca0052659, 3

[013130]<V> ###############IOTX_AWSS_GOT_SSID_PASSWD
[013380]<F> [chan scanning] 11340 ms

[013380]<F> final chan 2

[013380]<F> [chan recving] 11340 ms

[013380]<V> ################IOTX_AWSS_CONNECT_AHA
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
pm open phy_2,type:2 0 0
cnt 

connected with aha, channel 3
dhcp client start...
[WiFi] event 1
ip:192.168.43.207,mask:255.255.255.0,gw:192.168.43.1
[WiFi] event 4
[017020]<I> Got ip : 192.168.43.207, gw : 192.168.43.1, mask : 255.255.255.0
[017020]<I> In hotspot/router mode, do not post GOT_IP event here.
Wifi connected.
[017380]<I> AP connected
[017380]<F> awss connect ssid:aha success
[017380]<V> #################IOTX_AWSS_GOT_IP
#sun ##[awss_cmp_local_init]
awss register /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/device/switchap
###sun [CoAPServer_register]
awss register /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/event/wifilist/get
###sun [CoAPServer_register]
awss register /sys/awss/device/info/get
###sun [CoAPServer_register]
awss register /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/device/info/get
###sun [CoAPServer_register]
######sun [awss_notify_dev_info]
[017390]<D> topic:/sys/awss/device/info/notify

[017400]<I> topic:/sys/awss/device/info/notify

[017400]<D> payload:{"id":"1","version":"1.0","method":"awss.device.info.notify","params":{"awssVer":{"smartconfig":"2.0","zconfig":"2.0","router":"2.0","ap":"2.0","softap":"2.0"},"productKey":"a1YlsqdAPUS","deviceName":"pkZ8E2AH1f8NChrMKpiH","mac":"68:C6:3A:B2:6D:0B","ip":"192.168.43.207","cipherType":4,"random":"37E0C9EAF440C9AD46CFF85630F6B668","signMethod":0,"sign":"2B9A5D7C010233E2B0CDC9A3018BFCF8D32587D3"}}

[017400]<I> send notify success
[017410]<V> #################IOTX_AWSS_SETUP_NOTIFY
[017410]<D> --------------------------------AWSS STATIS-----------------------------------
[017410]<D> name                max     min     mean    cnt     suc     crc-err passwd-err
[017410]<D> Router              0       0       0       0       0       0       0
[017410]<D> SM                  0       0       0       0       0       0       0
[017410]<D> WPS                 0       0       0       0       0       0       0
[017410]<D> Aha                 0       0       0       1       0       0       0
[017410]<D> Adha                0       0       0       0       0       0       0
[017410]<D> Zconfig             0       0       0       0       0       0       0
[017410]<D> Dev-ap              0       0       0       0       0       0       0
[017410]<D> ------------------------------------------------------------------------------
[017410]<D> start, ssid:aha, strlen:3

#########aha get ip address 
#sun ##[awss_cmp_local_init]
awss register /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/device/switchap
###sun [CoAPServer_register]
[017420]<I> CoAPResource_register:Alread exist
[017420]<I> The resource /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/device/switchap already exist, re-write it
awss register /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/event/wifilist/get
###sun [CoAPServer_register]
[017430]<I> CoAPResource_register:Alread exist
[017430]<I> The resource /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/event/wifilist/get already exist, re-write it
awss register /sys/awss/device/info/get
###sun [CoAPServer_register]
[017430]<I> CoAPResource_register:Alread exist
[017430]<I> The resource /sys/awss/device/info/get already exist, re-write it
awss register /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/device/info/get
###sun [CoAPServer_register]
[017430]<I> CoAPResource_register:Alread exist
[017430]<I> The resource /sys/a1YlsqdAPUS/pkZ8E2AH1f8NChrMKpiH/awss/device/info/get already exist, re-write it
###sun[awss_devinfo_notify_resp]
[017490]<D> payload:{"id":1, "code":200, "data":{}}

[062020]<V> ###############IOTX_AWSS_ENALBE_TIMEOUT

xfdr0805, 2019-06-10:
你是用什么APP配网的，我用Iot Studio 生成的APP无法配网

jsonsiyuan, 2019-06-13:
问题已经解决，手机起热点配网，

closed 2022-03-01 by skylarCai