alibaba/AliOS-Things #997: 符号或者变量未定义，跳转至错误的同名头文件

athmoon, 2019-06-20:
**符号或者变量未定义**

**Win10  VScode  aos-cube version: 0.3.3 **
1. branch(version):  
2. Desktop OS(windows linux or macOS):Win10
3. example: 

**Expected behavior**
1，符号或者变量未定义
2，

**Screenshots**

[image]

 
[image]

**Additional context**

athmoon, 2019-06-20:
通过明确指定路径，将其他同名include路径排除掉应该就可以了
ctrl=Shift+P 
C/C++: Edit configurations (UI)
Include path 

${workspaceFolder}/platform/mcu/stm32f1xx/drivers/CMSIS/Device/ST/STM32F1xx/Include
${workspaceFolder}/3rdparty/**
${workspaceFolder}/app/example/blink/**
${workspaceFolder}/board/stm32f103ve-wsc/**
${workspaceFolder}/platform/mcu/stm32f1xx/drivers/STM32F1xx_HAL_Driver/Inc/**
${workspaceFolder}/drivers/**
${workspaceFolder}/include/**
${workspaceFolder}/kernel/**
${workspaceFolder}/middleware/**
${workspaceFolder}/network/**
${workspaceFolder}/osal/**
${workspaceFolder}/platform/mcu/stm32f1xx/**

xiaowenxia, 2019-06-20:
👍 赞！！

athmoon, 2019-06-20:
ctrl=Shift+P
C/C++: Edit configurations (UI)
Defines 根据芯片类型定义例如：
[image]

STM32F103xE

athmoon, 2019-06-20:
aos 如果可以自动配置 include和define就好了！

closed 2019-06-20 by athmoon