alibaba/AliOS-Things #1005: /usr/bin/bash: -d: invalid option

liubaiwu, 2019-07-06:
**Describe the bug**
编译提示/usr/bin/bash: -d: invalid option   开发工具使用的是 vs code 最新版本，该安装的环境软件都已经安装。

完整提示
> Executing task: aos make helloworld@esp8266 -c config <

/usr/bin/bash: -d: invalid option
Usage:  /usr/bin/bash [GNU long option] [option] ...
        /usr/bin/bash [GNU long option] [option] script-file ...
GNU long options:
        --debug
        --debugger
        --dump-po-strings
        --dump-strings
        --help
        --init-file
        --login
        --noediting
        --noprofile
        --norc
        --posix
        --protected
        --rcfile
        --restricted
        --verbose
        --version
        --wordexp
Shell options:
        -ilrsD or -c command or -O shopt_option         (invocation only)
        -abefhkmnptuvxBCHP or -o option
终端进程已终止，退出代码: 2

终端将被任务重用，按任意键关闭。

**environment**
1. branch(version): rel_2.1.0
2. Desktop OS(windows linux or macOS):windows
3. example: helooworld

**Expected behavior**

**Screenshots**

**Additional context**

xiaowenxia, 2019-07-07:
你是使用windows WSL 编译的么？

liubaiwu, 2019-07-07:
对 是的。 是不是 不能使用这边版本。需要使用完整版 用虚拟机安装的ubuntu系统 ？

liubaiwu, 2019-07-07:
请问 在windows  10 下面应该怎么进行编译。我也是按照官方配置流程来弄的。

liubaiwu, 2019-07-07:
我尝试过两种方式进行编译，一种是 vs code 进行编译，提示的是上面的问题。另外一种是我通过命令使用windows 的wsl  ubuntu系统进行安装环境进行编译，也不行。使用windows10 wsl 编译的时候 执行命令 aos make helloworld@esp8266 时  打印很长一段日志，下面是截取的最下面部分。
'est/testcase/certificate/certificate_test/Config.in:4:warning: ignoring unsupported character '
'est/testcase/certificate/certificate_test/Config.in:4:warning: ignoring unsupported character '
'est/testcase/certificate/Config.in:5:warning: ignoring unsupported character '
'est/testcase/certificate/Config.in:5:warning: ignoring unsupported character '
'est/testcase/Config.in:9:warning: ignoring unsupported character '
'est/Config.in:4:warning: ignoring unsupported character '
'est/yunit/Config.in:1:warning: ignoring unsupported character '
'est/yunit/Config.in:2:warning: ignoring unsupported character '
'est/yunit/Config.in:3:warning: ignoring unsupported character '
'est/yunit/Config.in:4:warning: ignoring unsupported character '
'est/yunit/Config.in:6:warning: ignoring unsupported character '
'est/yunit/Config.in:8:warning: ignoring unsupported character '
'est/yunit/Config.in:8:warning: ignoring unsupported character '
'est/Config.in:6:warning: ignoring unsupported character '
build/build_rules/aos_kconfig.mk:131: recipe for target '.config' failed
make: *** [.config] Error 1
[AliOS-Things] ERROR: "/mnt/i/esp8266/github/AliOS-Things/build/cmd/linux64/make" returned error code 2.
[AliOS-Things] ERROR: Command "/mnt/i/esp8266/github/AliOS-Things/build/cmd/linux64/make HOST_OS=Linux64 TOOLS_ROOT=/mnt/i/esp8266/github/AliOS-Things/build -e -f build/Makefile helloworld@esp8266.config" in "/mnt/i/esp8266/github/AliOS-Things"
---

closed 2019-07-07 by liubaiwu