alibaba/AliOS-Things #1010: win10环境下helloworld@developerkit编译无法通过

ArronChan, 2019-07-12:
点击左下角的编译按钮之后无法通过

[image]

报错信息为
```
> Executing task: aos make helloworld@developerkit -c config <

aos-cube version: 0.3.3
build/Makefile:288: target 'Help' given more than once in the same rule
process_begin: CreateProcess(NULL, git clone https://gitee.com/alios-things/kconfig-frontends-win32.git ./build/kconfig, ...) failed.
make (e=2): 系统找不到指定的文件。
make.exe: *** [build/build_rules/aos_kconfig.mk:60: .\build\kconfig\kconfig-conf.bat] Error 2
[AliOS-Things] ERROR: "C:/AliOS-Things-2.1.0/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "C:/AliOS-Things-2.1.0/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=C:/AliOS-Things-2.1.0/build -e -f build/Makefile helloworld@developerkit.config" in "C:\AliOS-Things-2.1.0"
---
终端进程已终止，退出代码: 1

终端将被任务重用，按任意键关闭。

```

xiaowenxia, 2019-07-12:
有安装[git](https://git-scm.com/downloads)嘛？

ArronChan, 2019-07-12:
@xiaowenxia 没有呢 需要安装吗？

xiaowenxia, 2019-07-12:
需要安装哦。

ArronChan, 2019-07-12:
> 需要安装哦。

好的 我现在安装

ArronChan, 2019-07-12:
解决了，需要下载的是branch下的developer，我之前用的都是rel

closed 2019-07-12 by ArronChan