alibaba/AliOS-Things #1017: rv32m1_v编译失败

niannianyouyu, 2019-07-29:
[image]
问题1：明明已经下载好toolchain，并放入指定位置，编译时却显示：toolchain missing
[image]
问题2：对此构架没有匹配的工具链，之前没有报toolchain missing时也出现了这个问题

xiaowenxia, 2019-07-29:
新的开发板需要在[build/toolchain_config.py](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/toolchain_config.py)中配置一下toolchain哈。

niannianyouyu, 2019-07-29:
> 可以在[build/toolchain_config.py](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/toolchain_config.py)中配置一下这个开发板的toolchain哈。

[image]

谢谢指导，这是我添加的部分，模仿其他板子写的，不知道命名规则有什么讲究，试了好久也还是不行

xiaowenxia, 2019-07-29:
可以不需要下载哈，直接把toolchain放到build/compiler/riscv32下面就行。

niannianyouyu, 2019-07-29:
> 1. `command`是toolchain的gcc名称，比如arm的工具链gcc名称为：`arm-none-eabi-gcc`，那command就要填`arm-none-eabi-gcc`。
> 2. 下载url一定要是git地址，`aos-cube`是调用`git clone`来下载toolchain的。
> 3. 也可以不需要下载哈，直接把toolchain放到`build/compiler/riscv32`下面就行（**目录结构要和arm-none-eabi-gcc的目录结构一样**）。aos-cube发现`build/compiler/riscv32`下面已经有toolchain的话，就直接用了。

[image]
哇，谢谢老哥，按你上一条回复的第1，3点，把本地的toolchain放入指定目录,`toolchain missing`的问题已经解决了，不过现在还是说 `No matching toolchain found for architecture`,是不是和`aos_target_config.mk`这个文件也有关系

niannianyouyu, 2019-07-29:
> 1. `command`是toolchain的gcc名称，比如arm的工具链gcc名称为：`arm-none-eabi-gcc`，那command就要填`arm-none-eabi-gcc`。
> 2. 下载url一定要是git地址，`aos-cube`是调用`git clone`来下载toolchain的。
> 3. [...]

[image]
[image]

这是`aos_target_config.mk`文件下的部分代码，`aos_toolchain_gcc.mk`文件里面已经包含了riscv32的
`aos_toolchain_riscv32_unkown_elf_gcc.mk`,这样还不行吗

xiaowenxia, 2019-07-29:
根据board下面的makefile中的HOST_ARCH字段看，使用的是[aos_toolchain_pulp_riscv32_unkown_elf_gcc.mk](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/build_rules/toolchain/aos_toolchain_pulp_riscv32_unkown_elf_gcc.mk)这个文件，在[aos_toolchain_gcc.mk](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/build_rules/toolchain/aos_toolchain_gcc.mk)中把这个文件include进去试一试哈。

niannianyouyu, 2019-07-29:
> 根据board下面的makefile中的HOST_ARCH字段看，使用的是[aos_toolchain_pulp_riscv32_unkown_elf_gcc.mk](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/build_rules/toolchain/aos_toolchain_pulp_riscv32_unkown [...]

[image]
[image]
好像还是不行，我的板子的platform/arch和platform/mcu下的Config.in，官方的代码里面没有，是我自己写的（如图），是不是有问题

xiaowenxia, 2019-07-29:
试试在[aos_toolchain_gcc.mk](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/build_rules/toolchain/aos_toolchain_gcc.mk)中增加这一行看看行不行：

```diff
index d2f93d0aa..abf4979af 100644
--- a/build/build_rules/toolchain/aos_toolchain_gcc.mk
+++ b/build/build_rules/toolchain/aos_toolchain_gcc.mk
@@ -6,4 +6,5 @@ include $(MAKEFILES_PATH)/toolchain/aos_toolchain_xtensa.mk
 include $(MAKEFILES_PATH)/toolchain/aos_toolchain_csky.mk
 include $(MAKEFILES_PATH)/toolchain/aos_toolchain_mips.mk
 include $(MAKEFILES_PATH)/toolchain/aos_toolchain_riscv32_unkown_elf_gcc.mk
+include $(MAKEFILES_PATH)/toolchain/aos_toolchain_pulp_riscv32_unkown_elf_gcc.mk
 include $(MAKEFILES_PATH)/toolchain/aos_toolchain_nds32le-elf-newlib-v3.mk
```

niannianyouyu, 2019-07-29:
> 试试在[aos_toolchain_gcc.mk](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/build_rules/toolchain/aos_toolchain_gcc.mk)中增加这一行看看行不行：
>
> ```diff
> index d2f93d0aa..abf4979af 100644
> --- [...]

我就是这么改的😭 @xiaowenxia

niannianyouyu, 2019-07-29:
> 试试在[aos_toolchain_gcc.mk](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/build/build_rules/toolchain/aos_toolchain_gcc.mk)中增加这一行看看行不行：
>
> ```diff
> index d2f93d0aa..abf4979af 100644
> --- [...]

[image]
我的编译结果architecture后面一直是空的，是不是因为HOST_ARCH没有接收到值

niannianyouyu, 2019-07-29:
的确是因为没有HOST_ARCH值，我在'board/riscv32_vega'目录下定义了aos.mk,文件里面定义了

xiaowenxia, 2019-07-29:
[rv32m1_vega.patch.zip](https://github.com/alibaba/AliOS-Things/files/3442479/rv32m1_vega.patch.zip)

打上这个patch试一试（解压要AliOS Things目录下，运行如下指令）：
```sh
git apply ./rv32m1_vega.patch
```

在linux上是没有问题的。

注意事项：
1. toolchain的存放位置如下（**注意路径名称**）：
```sh
$ tree -L 3 ./build/compiler/
./build/compiler/
└── riscv32-unknown-elf-gcc
    └── Linux64
        ├── bin
        ├── include
        ├── lib
        ├── libexec
        ├── riscv32-unknown-elf
        └── share
```
2. 这个toolchain需要安装依赖（参考[织女星开发板嵌入式开发环境搭建.pdf](https://github.com/open-isa-cn/vega-lite/blob/master/%E7%BB%87%E5%A5%B3%E6%98%9F%E5%BC%80%E5%8F%91%E6%9D%BF%E5%B5%8C%E5%85%A5%E5%BC%8F%E5%BC%80%E5%8F%91%E7%8E%AF%E5%A2%83%E6%90%AD%E5%BB%BA.pdf)）：

```sh
$ sudo apt install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libusb-1.0-0-dev libudev1 libudev-dev g++
```
3. 同时需要创建软符号才能正常编译：
```sh
$ sudo ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.4.1.4 /usr/lib/x86_64-linux-gnu/libmpfr.so.6
```

最终编译效果：

```
Making out/helloworld@rv32m1_vega/libraries/helloworld.a
Making out/helloworld@rv32m1_vega/libraries/board_rv32m1_vega.a
Making out/helloworld@rv32m1_vega/libraries/osal_aos.a
Making out/helloworld@rv32m1_vega/libraries/rhino.a
Making out/helloworld@rv32m1_vega/libraries/debug.a
Making out/helloworld@rv32m1_vega/libraries/kernel_init.a
Making out/helloworld@rv32m1_vega/libraries/arch_riscy.a
Making out/helloworld@rv32m1_vega/libraries/kv.a
Making out/helloworld@rv32m1_vega/libraries/mcu_rv32m1.a
Making out/helloworld@rv32m1_vega/libraries/netmgr.a
Making out/helloworld@rv32m1_vega/libraries/newlib_stub.a
Making out/helloworld@rv32m1_vega/libraries/ulog.a
Making out/helloworld@rv32m1_vega/libraries/vfs.a
Making out/helloworld@rv32m1_vega/libraries/yloop.a
Making helloworld@rv32m1_vega.elf

Making helloworld@rv32m1_vega.bin
Making helloworld@rv32m1_vega.hex
Can't parse memory configure, memory info get fail!
1048464
604f0e9b9f017e7b3077eea3cd4c2a5f
Build complete: helloworld@rv32m1_vega
```

niannianyouyu, 2019-07-29:
> [rv32m1_vega.patch.zip](https://github.com/alibaba/AliOS-Things/files/3442479/rv32m1_vega.patch.zip)
> 
> 打上这个patch试一试（解压要AliOS Things目录下，运行如下指令）：
> 
> ```shell
> git apply ./rv32m1_vega.patch
> ```
> 
> 在linux上是没有问题的。
> 
> 注意事项：
> 
> 1. toolchain的存放位置如下（**注意路径名称**）：
> 
> ```shell
> $ tree -L 3 ./build/compiler/
> ./build/compiler/
> └── riscv32-unknown-elf-gcc
>     └── Linux64
>         ├── bin
>         ├── include
>         ├── lib
>         ├── libexec
>         ├── riscv32-unknown-elf
>         └── share
> ```
> 
> 1. 这个toolchain需要安装依赖（参考[织女星开发板嵌入式开发环境搭建.pdf](https://github.com/open-isa-cn/vega-lite/blob/master/%E7%BB%87%E5%A5%B3%E6%98%9F%E5%BC%80%E5%8F%91%E6%9D%BF%E5%B5%8C%E5%85%A5%E5%BC%8F%E5%BC%80%E5%8F%91%E7%8E%AF%E5%A2%83%E6%90%AD%E5%BB%BA.pdf)）：
> 
> ```shell
> $ sudo apt install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libusb-1.0-0-dev libudev1 libudev-dev g++
> ```
> 
> 1. 同时需要创建软符号才能正常编译：
> 
> ```shell
> $ sudo ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.4.1.4 /usr/lib/x86_64-linux-gnu/libmpfr.so.6
> ```
> 
> 最终编译效果：
> 
> ```
> Making out/helloworld@rv32m1_vega/libraries/helloworld.a
> Making out/helloworld@rv32m1_vega/libraries/board_rv32m1_vega.a
> Making out/helloworld@rv32m1_vega/libraries/osal_aos.a
> Making out/helloworld@rv32m1_vega/libraries/rhino.a
> Making out/helloworld@rv32m1_vega/libraries/debug.a
> Making out/helloworld@rv32m1_vega/libraries/kernel_init.a
> Making out/helloworld@rv32m1_vega/libraries/arch_riscy.a
> Making out/helloworld@rv32m1_vega/libraries/kv.a
> Making out/helloworld@rv32m1_vega/libraries/mcu_rv32m1.a
> Making out/helloworld@rv32m1_vega/libraries/netmgr.a
> Making out/helloworld@rv32m1_vega/libraries/newlib_stub.a
> Making out/helloworld@rv32m1_vega/libraries/ulog.a
> Making out/helloworld@rv32m1_vega/libraries/vfs.a
> Making out/helloworld@rv32m1_vega/libraries/yloop.a
> Making helloworld@rv32m1_vega.elf
> 
> Making helloworld@rv32m1_vega.bin
> Making helloworld@rv32m1_vega.hex
> Can't parse memory configure, memory info get fail!
> 1048464
> 604f0e9b9f017e7b3077eea3cd4c2a5f
> Build complete: helloworld@rv32m1_vega
> ```
> 
> > `riscv32_vega` 这个board在最新的版本中默认不支持了。所以存在这些问题。

> [rv32m1_vega.patch.zip](https://github.com/alibaba/AliOS-Things/files/3442479/rv32m1_vega.patch.zip)
> 
> 打上这个patch试一试（解压要AliOS Things目录下，运行如下指令）：
> 
> ```shell
> git apply ./rv32m1_vega.patch
> ```
> 
> 在linux上是没有问题的。
> 
> 注意事项：
> 
> 1. toolchain的存放位置如下（**注意路径名称**）：
> 
> ```shell
> $ tree -L 3 ./build/compiler/
> ./build/compiler/
> └── riscv32-unknown-elf-gcc
>     └── Linux64
>         ├── bin
>         ├── include
>         ├── lib
>         ├── libexec
>         ├── riscv32-unknown-elf
>         └── share
> ```
> 
> 1. 这个toolchain需要安装依赖（参考[织女星开发板嵌入式开发环境搭建.pdf](https://github.com/open-isa-cn/vega-lite/blob/master/%E7%BB%87%E5%A5%B3%E6%98%9F%E5%BC%80%E5%8F%91%E6%9D%BF%E5%B5%8C%E5%85%A5%E5%BC%8F%E5%BC%80%E5%8F%91%E7%8E%AF%E5%A2%83%E6%90%AD%E5%BB%BA.pdf)）：
> 
> ```shell
> $ sudo apt install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libusb-1.0-0-dev libudev1 libudev-dev g++
> ```
> 
> 1. 同时需要创建软符号才能正常编译：
> 
> ```shell
> $ sudo ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.4.1.4 /usr/lib/x86_64-linux-gnu/libmpfr.so.6
> ```
> 
> 最终编译效果：
> 
> ```
> Making out/helloworld@rv32m1_vega/libraries/helloworld.a
> Making out/helloworld@rv32m1_vega/libraries/board_rv32m1_vega.a
> Making out/helloworld@rv32m1_vega/libraries/osal_aos.a
> Making out/helloworld@rv32m1_vega/libraries/rhino.a
> Making out/helloworld@rv32m1_vega/libraries/debug.a
> Making out/helloworld@rv32m1_vega/libraries/kernel_init.a
> Making out/helloworld@rv32m1_vega/libraries/arch_riscy.a
> Making out/helloworld@rv32m1_vega/libraries/kv.a
> Making out/helloworld@rv32m1_vega/libraries/mcu_rv32m1.a
> Making out/helloworld@rv32m1_vega/libraries/netmgr.a
> Making out/helloworld@rv32m1_vega/libraries/newlib_stub.a
> Making out/helloworld@rv32m1_vega/libraries/ulog.a
> Making out/helloworld@rv32m1_vega/libraries/vfs.a
> Making out/helloworld@rv32m1_vega/libraries/yloop.a
> Making helloworld@rv32m1_vega.elf
> 
> Making helloworld@rv32m1_vega.bin
> Making helloworld@rv32m1_vega.hex
> Can't parse memory configure, memory info get fail!
> 1048464
> 604f0e9b9f017e7b3077eea3cd4c2a5f
> Build complete: helloworld@rv32m1_vega
> ```
> 
> > `riscv32_vega` 这个board在最新的版本中默认不支持了。所以存在这些问题。

[image]

十分感谢，不过还是比较好奇Windows上的错误，之前提到的问题我已经解决了，现在kv.c报错，您能看出这是什么原因造成的吗

niannianyouyu, 2019-07-30:
linux下板子按照如上的步骤，应该是可以用了；Windows下编译还是报错，水平有限，有机会再研究吧

closed 2019-07-30 by niannianyouyu