alibaba/AliOS-Things #1027: 请问 庆科3080 不能进入GPIO中断

xiaoyaodian, 2019-08-14:
使用EWM3080 2.1.0alithing 分支 GPIO 输入输出都正常但是进不来GPIO口中断服务函数，要怎么做，是hal文件没有适配吗？platform\mcu\下也找不到MK3080的适配文件
[image]

xiaowenxia, 2019-08-14:
mk3080用的mcu是[rtl8710bn](https://github.com/alibaba/AliOS-Things/tree/rel_2.1.0/platform/mcu/rtl8710bn)，gpio的hal代码是适配了的。

xiaoyaodian, 2019-08-14:
@xiaowenxia 可是，GPIO进不了中断服务函数啊，   hal_gpio_enable_irq(&input, IRQ_TRIGGER_BOTH_EDGES, gpio_isr_handler, (void *) MICO_GPIO_7);

xiaoyaodian, 2019-08-14:
编译的时候会自动适配rtl8710bn下的hal函数吗？

xiaowenxia, 2019-08-14:
[image]
看源码，`IRQ_TRIGGER_BOTH_EDGES`并没有作用，你换成`IRQ_TRIGGER_FALLING_EDGE`或者`IRQ_TRIGGER_RISING_EDGE`试一次。

同样，`hal_gpio_init()`好像也对`IRQ_MODE`的gpio模式没有作用，换成其他的试一试。

xiaoyaodian, 2019-08-14:
[image]

xiaoyaodian, 2019-08-14:
@xiaowenxia 大佬，这样写还是不行啊

xiaowenxia, 2019-08-14:
你参考下mk3080（`board/mk3080/board.c`）的配网按键的[相关代码](https://github.com/alibaba/AliOS-Things/blob/rel_2.1.0/board/mk3080/board.c#L60)，也是使用gpio中断功能。

closed 2022-03-01 by skylarCai