alibaba/AliOS-Things #1028: [ESP8266][天猫精灵] 如何使用 ESP8266, 基于 AliOS, 连接天猫精灵

ustccw, 2019-08-14:
希望提供较为完善的文档，指导开发者完成 `ESP8266` 上 `AliOS` 连接天猫精灵。

xiaowenxia, 2019-08-14:
你具体是想怎么连接天猫精灵呢？目前AliOS Things支持2种方式对接天猫精灵：

* [通过BLE连接天猫精灵](https://github.com/alibaba/AliOS-Things/tree/rel_2.1.0/app/example/bluetooth/blemesh_tmall)。
* 接入[飞燕平台](https://living.aliyun.com)，启动飞燕平台的[天猫精灵服务](https://living.aliyun.com/doc#TmallGenie.html)。

ustccw, 2019-08-14:
想通过天猫精灵平台: https://iot.aligenie.com/home 创建产品.
通过天猫精灵音箱和天猫精灵 `APP` 控制.

xiaowenxia, 2019-08-14:
接入飞燕平台，启动飞燕平台的天猫精灵服务，也可以实现天猫精灵和天猫精灵app控制哈。

如果你想接入天猫精灵平台的话，需要使用天猫精灵平台的SDK来适配设备。

ustccw, 2019-08-14:
这里的 alios 版本似乎可以连接天猫精灵, 不过没有详细的文档说明。 https://doc-bot.tmall.com/docs/doc.htm?spm=0.7629140.0.0.393b1780uuDN6X&treeId=578&articleId=118056&docType=1

xiaowenxia, 2019-08-14:
这个功能和使用方法跟AliOS Things类似，只是接入到的是天猫精灵平台。

xuhongv, 2019-08-24:
图文并茂教你使用乐鑫esp8266轻松连接阿里飞燕平台，个人设备实现轻松对接天猫精灵 。( - CSDN博客 https://blog.csdn.net/xh870189248/article/details/84958871

@ustccw  参考这个就可以了，只要跑起来 `linkkitapp` 就可以了！一年前我就研究出来了，没有途径去推广；esp8266确实是个很强大的，这个支持个人开发，意味大学生都可以在校做；

xuhongv, 2019-08-24:
esp8266通过aliosthings操作系统连接阿里的飞燕平台，实现天猫精灵控制。

angle600, 2019-10-01:
> 你具体是想怎么连接天猫精灵呢？目前AliOS Things支持2种方式对接天猫精灵：
> 
> * [通过BLE连接天猫精灵](https://github.com/alibaba/AliOS-Things/tree/rel_2.1.0/app/example/bluetooth/blemesh_tmall)。
> * 接入[飞燕平台](https://living.aliyun.com)，启动飞燕平台的[天猫精灵服务](https://living.aliyun.com/doc#TmallGenie.html)。

wifi设备是否可以直连精灵云？

xuhongv, 2019-10-01:
- 【阿里云生活物联网架构师专题 ②】esp8266 sdk 直连接入阿里云物联网平台，实现天猫精灵找队友零配网功能和语音控制； - 徐宏的博客。 - CSDN博客 https://blog.csdn.net/xh870189248/article/details/101448052

xuhongv, 2019-10-01:
@angle600

panguoxian, 2019-11-30:
第一次使用Alios也是想接入aligenie，看了阿里一大堆平台，好乱。

closed 2022-03-01 by skylarCai