alibaba/AliOS-Things #1030: 8266 配置两个引脚中断及以上的引脚中断时异常

jkl82, 2019-09-01:
**Describe the bug**
8266 配置两个引脚中断，只有最后配的引脚测试通过，在它之前的引脚都变成最后一个引脚的中断了。 但只有一个引脚中断时就不会有问题

**environment**
AliOS Things
1. branch(version): 
2.1.0
2. Desktop OS(windows linux or macOS):
windows
3. example: 
linkkitapp
**Expected behavior**

**Screenshots**

**Additional context**

xiaowenxia, 2019-09-04:
代码可以贴上来看看哈。

jkl82, 2019-09-04:
以下请帮忙看看,谢谢！
gpio_dev_t LED1, LED2, LED3, SW1, SW2, SW3;
int LED1_status=1,LED2_status=1,LED3_status=1;//0亮，1灭
#define GPIO_LED1 13
#define GPIO_LED2 12
#define GPIO_LED3 16
#define GPIO_SW1 4
#define GPIO_SW2 5
#define GPIO_SW3 14

static void gpio_isr_handler_SW1(void *arg)
{
    int count=0;
    int lightswitch;
    uint32_t SW_value;
    hal_gpio_disable_irq(&SW1);        
        
        EXAMPLE_TRACE("IRQ is running SW1!\n");
        
        if(LED1_status==1)
        {
            LED1_status=0;
            hal_gpio_output_low(&LED1);
        }
        else
        {
            LED1_status=1;
            hal_gpio_output_high(&LED1);
        }
       hal_gpio_clear_irq(&SW1);
    
    return ;    
}

static void gpio_isr_handler_SW2(void *arg)
{
    int count=0;
    int lightswitch;
    uint32_t SW_value;  
        
    hal_gpio_disable_irq(&SW2);
    
    EXAMPLE_TRACE("IRQ is running SW2!\n");
        
        if(LED2_status==1)
        {
            LED2_status=0;
            hal_gpio_output_low(&LED2);
        }
        else
        {
            LED2_status=1;
            hal_gpio_output_high(&LED2);
        }
       
       hal_gpio_clear_irq(&SW2);

    return ;    
}

void GPIO_Init()
{
    int ret = -1;
    /* gpio port config */
    LED1.port = GPIO_LED1;
    /* set as output mode */
    LED1.config = OUTPUT_PUSH_PULL;
    /* configure GPIO with the given settings */
    hal_gpio_init(&LED1);
    hal_gpio_output_high(&LED1); //高电平，关闭LED

    LED2.port = GPIO_LED2;
    LED2.config = OUTPUT_PUSH_PULL;
    hal_gpio_init(&LED2);
    hal_gpio_output_high(&LED2); //高电平,关闭LED

    /* SW1 pin config*/
    SW1.port = GPIO_SW1;
    // set as interrupt mode /
    SW1.config = IRQ_MODE; 

    //configure GPIO with the given settings /
    hal_gpio_init(&SW1);
    // gpio interrupt config,上升沿触发中断  /
    hal_gpio_enable_irq(&SW1, IRQ_TRIGGER_RISING_EDGE, gpio_isr_handler_SW1, (void *)GPIO_SW1);        
 
    // SW2 pin config 
    SW2.port = GPIO_SW2;    
    SW2.config = IRQ_MODE;
    hal_gpio_init(&SW2);
    hal_gpio_enable_irq(&SW2, IRQ_TRIGGER_RISING_EDGE, gpio_isr_handler_SW2, (void *)GPIO_SW2);

}
不管按下SW1还是SW2，一旦有按下，从串口中不管打印出（本以为接个上拉电阻就行，不过还是不行，只取一个按键中断就可以）：
[1;32;40mgpio_isr_handler_SW2.112: IRQ is running SW2!

closed 2022-03-01 by skylarCai