alibaba/AliOS-Things #1038: 按照文档搭建WIN10环境,编译报错，make路径错误，

shibaoyi, 2019-09-15:
按照文档搭建WIN10环境,编译报错，make路径错误，
*> Executing task: aos make helloworld@esp32devkitc <

aos-cube version: 0.3.5
Check if required tools for esp32devkitc exist

Build command updated, please follow steps:
*********************************************************************
Step 1: config:
$ aos make helloworld@esp32devkitc -c config
---> Or more config: $ aos make menuconfig

Step 2: build:
$ aos make    # VAR=value if needed

---> Or oneline command for config and build:
$ aos make helloworld@esp32devkitc -c config && aos make
*********************************************************************
build/Makefile:119: *** Build Target Deprecated!.  Stop.
[AliOS-Things] ERROR: "d:/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "d:/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=d:/AliOS-Things/build -e -f build/Makefile helloworld@esp32devkitc" in "d:\AliOS-Things"
---

andyjefflin, 2019-09-19:
我在ubuntu和win7环境下都出现这个问题，也换过电脑重装过系统，也是这个问题，很头疼。

angle600, 2019-09-29:
我也一样出现这个问题，但macOS没有出现，头疼

ECHO 处于关闭状态。
Build Configuration:
====================
App:   linkkitapp
Board: esp8266
====================
ECHO 处于关闭状态。
Creating out/config/autoconf.h ...
Making config file for first time
app_version:app-1.0.0-20190929.183415
processing components: linkkitapp esp8266
*** All Components: linkkitapp esp8266 linkkit_sdk_c netmgr cjson ota mcu_esp8266 kernel_init libiot_wrappers libiot_coap libiot_devmodel libiot_sign libiot_mqtt libiot_awss kv osal_aos yloop http mbedtls newlib_stub debug lwip rhino arch_xtensa_lx106 ulog libiot_infra activation chip_code libiot_certs vfs lib_rbtree
*** Config Enabled: lib_rbtree
make.exe[1]: write error: stdout
make.exe: *** [build/Makefile:337: out/linkkitapp@esp8266/config.mk] Error 1
[AliOS-Things] ERROR: "D:/work/esp/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "D:/work/esp/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=D:/work/esp/AliOS-Things/build -e -f build/Makefile" in "D:\work\esp\AliOS-Things"
---
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

xiaowenxia, 2019-09-29:
> 按照文档搭建WIN10环境,编译报错，make路径错误，
> *> Executing task: aos make helloworld@esp32devkitc <
>
> aos-cube version: 0.3.5
> Check if required tools for esp32devkitc exist
>
> Build command updated, please follow [...]

AliOS Things 2.1版本的代码，编译指令更改成了先配置，后编译的模式：
```sh
# 配置
$ aos make helloworld@esp32devkitc -c config
# 编译
$ aos make
```

xiaowenxia, 2019-09-29:
> 我也一样出现这个问题，但macOS没有出现，头疼
>
> # ECHO 处于关闭状态。
> Build Configuration:
> # App: linkkitapp
> Board: esp8266
> ## ECHO 处于关闭状态。
> Creating out/config/autoconf.h ...
> Making config file for first time [...]

你这个看起来是make 指令，stdout标准输出有问题，试试关掉系统防火墙？

angle600, 2019-09-29:
> 你这个看起来是make 指令，stdout标准输出有问题，试试关掉系统防火墙？

没有防火墙

panguoxian, 2019-12-20:
我也有一样的问题，不知如何解决。。。哭哭

LinTeX9527, 2020-01-03:
一样的问题，aos-cube 安装了两次，一次是vscode一键安装，一次是手动安装，都没有解决这个问题。

closed 2022-03-01 by skylarCai