alibaba/AliOS-Things #1039: ERROR: Import scons_upload.py: failed No module named scons_upload

ThisIsMarxMA, 2019-09-16:
why would this happen when i trying to upload to board? with python-conda 2.7.16

ThisIsMarxMA, 2019-09-16:
this file scons_upload.py clearly exist in folder AliOS-Things/site_scons.

KtQiu, 2019-10-08:
You can try
`pip install --upgrade setuptools
pip install --upgrade wheel`

FeiBoNaQi, 2019-10-11:
have you solved it? I encounter the same problem

xiaowenxia, 2019-10-11:
> have you solved it? I encounter the same problem

AliOS Things什么版本呢？ 1.3.x版本不支持aos upload命令。

FeiBoNaQi, 2019-10-12:
我解决了，是因为这个文件没有在python的路径里，我在我的site-package里面加入了一个.pth文件，然后把site_scons文件夹的地址放进去，就OK了。
但是我遇到了另一个问题，下面是我的命令行：
> Executing task: aos upload mqttapp@stm32f412zg-nucleo <

aos-cube version: 0.3.7
[INFO]: Currently in aos_sdk_path: '/Users/feibonaqi/Downloads/aiot/ST_412'
[INFO]: Target: mqttapp@stm32f412zg-nucleo
Traceback (most recent call last):
  File "/anaconda3/envs/aiot/lib/python2.7/site-packages/aos/__main__.py", line 83, in main
    cli(None)
  File "/anaconda3/envs/aiot/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/anaconda3/envs/aiot/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/anaconda3/envs/aiot/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/anaconda3/envs/aiot/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/anaconda3/envs/aiot/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/anaconda3/envs/aiot/lib/python2.7/site-packages/aos/commands/upload.py", line 97, in cli
    aos_upload(upload_target, work_path, binaries_dir)
TypeError: aos_upload() takes exactly 1 argument (3 given)
[AliOS-Things] ERROR: Unknown Error: aos_upload() takes exactly 1 argument (3 given)
---
The terminal process terminated with exit code: 255
请问有哪位大哥能帮忙看看吗？然后我的AliOS Things插件是刚下载的，版本应该够

ThisIsMarxMA, 2019-10-18:
TypeError: aos_upload() takes exactly 1 argument (3 given)
根据这个报错情况看的话是函数参数多了
解决办法就是直接简单粗暴的改掉...
找到site_packages/aos/commands/upload.py : 第97行
注释掉 #aos_upload(upload_target, work_path, binaries_dir) 
改成 aos_upload(upload_target)
然后我的就上传成功了...

closed 2022-03-01 by skylarCai