alibaba/AliOS-Things #1042: win10 环境下 vscode 编译 hello-world 工程错误

Lawlieta, 2019-09-19:
aos-cube 版本 0.3.6
python 版本 2.7.14

win10 环境下 vscode 编译 hello-world 工程错误，日志如下：
``` shell
$ git clone https://gitee.com/alios-things/kconfig-frontends-win32.git ./build/kconfig/Win32/

make.exe: *** [build/build_rules/aos_kconfig.mk:94: .\build\kconfig\Win32\kconfig-conf.bat] Error 1
[AliOS-Things] ERROR: "E:/RT-Thread/AliOS-Things/build\cmd/win32/make.exe" returned
error code 2.
[AliOS-Things] ERROR: Command "E:/RT-Thread/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/RT-Thread/AliOS-Things/build -e -f build/Makefile helloworld@stm32f429zi-nucleo.config" in "E:\RT-Thread\AliOS-Things"
---
终端进程已终止，退出代码: 1

终端将被任务重用，按任意键关闭。
```

麻烦帮忙分析一下什么原因？？

closed 2019-09-19 by Lawlieta

canberry001, 2019-09-29:
可以说下什么原因吗？怎么解决的？

xiaowenxia, 2019-09-29:
手动运行这条指令就好了：

```sh
git clone https://gitee.com/alios-things/kconfig-frontends-win32.git ./build/kconfig/Win32/
```