alibaba/AliOS-Things #1043: 关于ble mesh的疑问

wangmengyuan2, 2019-09-19:
请问目前aliosthings对于ble mesh的支持是什么状态，文档README里面支持的pca10056验过吗？

xiaowenxia, 2019-10-11:
pca10056是验证过的。

iiiilii, 2019-10-18:
> pca10056是验证过的。

请问3.0编译报错如何解决？如下

python.exe: can't open file './platform/mcu/moc108//padzero.py': [Errno 2] No such file or directory
make.exe[1]: *** [platform/mcu/moc108//gen_image_bin.mk:35: gen_standard_images] Error 2
make.exe: *** [build/Makefile:393: main_app] Error 2
[AliOS-Things] ERROR: "E:/AliOS-Things/AliOS-Things-rel_3.0.0/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "E:/AliOS-Things/AliOS-Things-rel_3.0.0/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=E:/AliOS-Things/AliOS-Things-rel_3.0.0/build -e -f build/Makefile" in "E:\AliOS-Things\AliOS-Things-rel_3.0.0"

WenzongFan, 2019-10-18:
从出错的mcu看，你编的应该是mk3060，先aos make distclean，再重新配置、编译试试

iiiilii, 2019-10-18:
> 从出错的mcu看，你编的应该是mk3060，先aos make distclean，再重新配置、编译试试

一样的

wangmengyuan2, 2019-10-18:
> pca10056是验证过的。

请问可以提供ble mesh的demo吗?

closed 2022-03-01 by skylarCai