alibaba/AliOS-Things #1045: linkkitapp@esp8266   ESP-12F启动出错

angle600, 2019-09-28:
我在macOS上用alios-studio烧录程序linkkitapp进ESP-12F，启动出错，请问应该如何修改呢？

2nd boot version : 1.7(5d6f877)

SPI Speed : 40MHz

SPI Mode : QIO

SPI Flash Size & Map: 32Mbit(512KB+512KB)

jump to run user1 @ 1000

!!!!!!!!!! Exception  !!!!!!!!!!

========== Regs info  ==========

PC       0x4024B281

PS       0x00000030

A0       0x4020F4B8

A1       0x3FFFF960

A2       0x00000000

A3       0x40238297

A4       0x000000FF

A5       0x0000FF00

A6       0x00FF0000

A7       0xFF000000

A8       0x000A726F

A9       0x00000002

A10      0x00000000

A11      0x00000102

A12      0x3FFE87EC

A13      0x00000014

A14      0x3FFFF9B0

A15      0x3FFFF980

SAR      0x00000016

EXCCAUSE 0x0000001C

EXCVADDR 0x00000010

========== Stack info ==========

stack(0x3FFFF960): 0x40103074 0x91303081 0x0000F00D 0x40103084 

stack(0x3FFFF970): 0x00F00D03 0x00000000 0x00000000 0x00000000 

stack(0x3FFFF980): 0x00000000 0x00000001 0x00000000 0x55AA55AB 

stack(0x3FFFF990): 0x3FFFF9E0 0x3FEFFE00 0x00000000 0x00000000 

stack(0x3FFFF9A0): 0x40238284 0x00000000 0x0000048C 0x40100590 

stack(0x3FFFF9B0): 0x00000000 0x00000000 0x00000000 0x00000020

xiaowenxia, 2019-09-29:
可以参考下这几个issues：
* #931
* #766 
* #507

angle600, 2019-09-30:
Makefile中加入：
BOOT?=new
APP?=1
SPI_SPEED?=40
SPI_MODE?=DOUT
SPI_SIZE_MAP?=6

esp8266.json中写入
{
    "cmd": [
        {
            "Linux32": "esptool.py",
            "Linux64": "esptool.py",
            "OSX": "esptool.py",
            "Win32": "esptool.exe"
        },
        "--chip",
        "esp8266",
        "--port",
        "@PORT@",
        "--baud",
        "921600",
        "write_flash",
        "--flash_mode",
        "dout",
        "--flash_size",
        "4MB",
        "0x0",
        "@AOSROOT@/platform/mcu/esp8266/bsp/boot_v1.7_921600.bin",
        "0x3fc000",
        "@AOSROOT@/platform/mcu/esp8266/bsp/esp_init_data_default.bin",
        "0x3fe000",
        "@AOSROOT@/platform/mcu/esp8266/bsp/blank.bin",
        "0x1000",
        "@AOSROOT@/out/@TARGET@/binary/@TARGET@.bin"
    ]
}

成功启动

closed 2019-09-30 by angle600