alibaba/AliOS-Things #1046: win10下aliOS-Studio如何更改shell为mingw32

angle600, 2019-09-29:
选择默认shell没有选项mingw32，这个应该如何设置呢？

xiaowenxia, 2019-09-29:
在你的系统settings.json中配置内置terminal就行，可以选择配置下面任意一种terminal：
```json
// Command Prompt
"terminal.integrated.shell.windows": "C:\\Windows\\System32\\cmd.exe"
// PowerShell
"terminal.integrated.shell.windows": "C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\powershell.exe"
// Git Bash
"terminal.integrated.shell.windows": "C:\\Program Files\\Git\\bin\\bash.exe"
// Bash on Ubuntu (on Windows)
"terminal.integrated.shell.windows": "C:\\Windows\\System32\\bash.exe"
```

更多说明可以参考vscode官方文档：[Intergrated-Terminal Configuration](https://code.visualstudio.com/docs/editor/integrated-terminal#_configuration)。

angle600, 2019-09-30:
> 在你的系统settings.json中配置内置terminal就行，可以选择配置下面任意一种terminal：
>
> ```json
> // Command Prompt
> "terminal.integrated.shell.windows": "C:\\Windows\\System32\\cmd.exe"
> // PowerShell [...]

设置了"terminal.integrated.shell.windows": "D:\\Program Files\\msys32\\mingw32.exe",但default Shell仍然为cmd.exe，只是新弹出一个mingw32窗口，task仍在cmd.exe里运行

xiaowenxia, 2019-10-01:
`D:\Program Files\msys32\mingw32.exe`应该不是shell执行程序吧？看看安装目录下面是否有bash.exe或者sh.exe`？
参考这篇文档看看：[Using MSYS2 MinGW 64-bit terminal on VSCode](https://stackoverflow.com/questions/53943308/using-msys2-mingw-64-bit-terminal-on-vscode)

angle600, 2019-10-04:
> `D:\Program Files\msys32\mingw32.exe`应该不是shell执行程序吧？看看安装目录下面是否有bash.exe或者sh.exe`？
> 参考这篇文档看看：[Using MSYS2 MinGW 64-bit terminal on VSCode](https://stackoverflow.com/questions/53943308/using-msys2-mingw-64-bit-terminal-on-vscode)

bin下有bash.exe，谢谢

closed 2019-10-04 by angle600