alibaba/AliOS-Things #1047: ESP8266无法配网

angle600, 2019-10-01:
sdk使用最新的master版本
aos make menuconfig开启console
ESP8266启动后在串口输入active_awss start
使用云智能、天猫精灵、IOT_Espressif_EspTouch均无法配网
但是却不断扫描，没有配置网络信息
这个应该如何解决呢？

active_awss start

active_awss start
[ 228.820]<A>AOS do_awss_active 1

[ 228.820]<F>awss enable awss

[ 228.830]<A>AOS IOTX_AWSS_ENABLE

# 
# [ 228.960]<F>awss chan 1

[ 229.230]<F>awss chan 6

[ 229.500]<F>awss chan 11

[ 229.770]<F>awss chan 1

[ 230.040]<F>awss chan 2

[ 230.310]<F>awss chan 3

[ 230.580]<F>awss chan 4

[ 230.850]<F>awss chan 5

[ 231.120]<F>awss chan 6

[ 231.390]<F>awss chan 7

[ 231.660]<F>awss chan 8

[ 231.930]<F>awss chan 9

[ 232.200]<F>awss chan 10

[ 232.470]<F>awss chan 11

[ 232.740]<F>awss chan 12

[ 233.010]<F>awss chan 13

[ 233.280]<F>awss chan 1

[ 233.550]<F>awss chan 6

[ 233.820]<F>awss chan 11

[ 234.090]<F>awss chan 13

[ 234.360]<F>awss chan 1

[ 234.630]<F>awss chan 6

[ 234.900]<F>awss chan 11

[ 235.170]<F>awss chan 1

[ 235.440]<F>awss chan 2

[ 235.710]<F>awss chan 3

[ 235.980]<F>awss chan 4

[ 236.250]<F>awss chan 5

[ 236.520]<F>awss chan 6

[ 236.790]<F>awss chan 7

[ 237.060]<F>awss chan 8

[ 237.330]<F>awss chan 9

[ 237.600]<F>awss chan 10

[ 237.870]<F>awss chan 11

[ 238.140]<F>awss chan 12

[ 238.410]<F>awss chan 13

[ 238.680]<F>awss chan 1

[ 238.950]<F>awss chan 6

[ 239.220]<F>awss chan 11

[ 239.300]<F>awss [7] ssid:WIFI_DVR_EFAD, mac:ac83f3ca3fc0, chn:11, rssi:-89

[ 239.490]<F>awss chan 13

[ 239.760]<F>awss chan 1

[ 240.030]<F>awss chan 6

angle600, 2019-10-01:
用天猫精灵扫描配网二维码，密码错误，三元组是在aligenie.com里申请的

[ 238.140]<F>awss 

65 65 65 65 64 64 64 64 
64 64 64 64 64 64 64 64 
65 64 64 64 64 64 64 64 
65 
[ 238.140]<F>awss 

19 0f 05 0e 5a 48 45 4e 
47 19 32 3d 20 07 25 24 
1e 1f 29 12 15 03 01 0d 
01 
[ 238.150]<F>awss 

[ 238.150]<F>awss SSID1: [zheng]

[ 238.160]<F>awss passwd err

[ 238.160]<A>AOS IOTX_AWSS_PASSWD_ERR
[ 238.160]<F>awss hint frame: offset:52, T, sn:b5b

[ 238.170]<F>awss src:f460e26f6f64, bssid:c061187f704f

[ 238.170]<A>AOS IOTX_AWSS_LOCK_CHAN
[ 238.180]<F>awss hint frame: offset:52, F, sn:58e

[ 238.180]<F>awss src:f460e26f6f64, bssid:c061187f704f

[ 238.190]<A>AOS IOTX_AWSS_LOCK_CHAN
[ 238.190]<F>awss 24+0 [100] b5c T 3e3

[ 238.190]<A>AOS IOTX_AWSS_LOCK_CHAN
[ 238.200]<F>awss 24+0 [100] 590 F 3e3

[ 238.200]<A>AOS IOTX_AWSS_LOCK_CHAN
[ 238.260]<F>awss 24+1 [100] b5d T 101

[ 238.350]<F>awss 24+1 [98] 592 F 101

[ 238.350]<F>awss 			 uplimit [98] -> [99]

xiaowenxia, 2019-10-01:
* esp8266因为RAM小的原因，禁用了CLI功能，所以无法输入active_awss start。
* 云智能app对esp8266配网，需要设置 四元组，而不是三元组。
* esp8266 请使用按键触发进入配网功能。

closed 2019-10-02 by angle600