alibaba/AliOS-Things #1051: ESP8266配网密码解密出错

angle600, 2019-10-02:
我在https://www.aligenie.com申请了4元组，在程序中设置好。用天猫精灵扫描配网二维码，程序中的解密方式为SEC_LVL_AES128_PRODUCT，开启了console，在cli中输入“devinfo get”，product id正确，输入“active_awss start"触发配网，却发现解密错误，请问应该如何解决呢？

[  53.780]<A>AOS IOTX_AWSS_LOCK_CHAN
[  53.780]<F>awss 

64 65 65 64 65 65 65 65 
65 65 64 64 5c 0a 54 54 
65 64 64 64 64 64 64 64 
65 
[  53.790]<F>awss 

19 0f 05 0e 5a 48 45 4e 
47 19 32 3d 20 07 25 24 
1e 1f 29 12 15 03 01 0d 
01 
[  53.790]<F>awss 

[  53.800]<F>awss SSID1: [zheng]

[  53.800]<F>awss passwd err

[  53.800]<A>AOS IOTX_AWSS_PASSWD_ERR
[  53.860]<F>awss hint frame: offset:52, T, sn:8e8

[  53.860]<F>awss src:f460e26f6f64, bssid:c061187f704f

[  53.860]<A>AOS IOTX_AWSS_LOCK_CHAN
[  53.880]<F>awss 16+0 [100] 8e9 T 3e2

[  53.880]<A>AOS IOTX_AWSS_LOCK_CHAN
[  53.900]<F>awss hint frame: offset:52, F, sn:dc2

[  53.900]<F>awss src:f460e26f6f64, bssid:c061187f704f

[  53.900]<A>AOS IOTX_AWSS_LOCK_CHAN
[  53.910]<F>awss 16+1 [100] 8ea T 11e

xiaowenxia, 2019-10-11:
确认下4元组应该从 https://www.aligenie.com 上申请， 还是从[生活物联网平台](https://living.aliyun.com/#/)上申请？

angle600, 2019-10-11:
> https://www.aligenie.com

https://www.aligenie.com 上申请的

xiaowenxia, 2019-10-11:
代码里面要设置的是4元组，包括`PRODUCT SECRET`。

FireWinter, 2019-10-12:
我从生活物联网平台申请的四元组，配置之后，alios-things 2.0及以下的版本配网OK，3.0及master版本配网过程解password会出错，应该是设备端代码解析问题

angle600, 2019-10-12:
> 我从生活物联网平台申请的四元组，配置之后，alios-things 2.0及以下的版本配网OK，3.0及master版本配网过程解password会出错，应该是设备端代码解析问题

啊？我是在aligence.com上申请的，官方也不说明3.0不能用。我不明白阿里为什么不用乐鑫的公版smartconfig无加密配网方式，这个加密配网对于安全性来说没有一点提高。微信的airkiss也没有加密，所有设备都可以用，只是配对的时候要验证身份。

FireWinter, 2019-10-12:
> > 我从生活物联网平台申请的四元组，配置之后，alios-things 2.0及以下的版本配网OK，3.0及master版本配网过程解password会出错，应该是设备端代码解析问题
> 
> 啊？我是在aligence.com上申请的，官方也不说明3.0不能用。我不明白阿里为什么不用乐鑫的公版smartconfig无加密配网方式，这个加密配网对于安全性来说没有一点提高。微信的airkiss也没有加密，所有设备都可以用，只是配对的时候要验证身份。

你可以考虑用espressif的阿里云：https://github.com/espressif/esp-aliyun.git，也支持天猫精灵

closed 2022-03-01 by skylarCai